vitotai/BrewPiLess #302: New to BrewPiLess trouble ds18b20

Nauticno, 2021-01-08:
Hi!
I am struggeling with ds18b20, at last I found a pair that worked. Then I wanted to extend the cable with rj45 using https://www.instructables.com/DS18B2O-Temperature-probe-with-RJ45-connector/ . My sensors are detected and I am able to apply them. Then the relay starts to click. After applying releays the clicking is getting weaker, but still clicking. When I click on Graph label I have no readings. What can be the issue?

daghovland, 2021-01-11:
You provide too few details. Relays click when they go on and off. There can be several reasons for this, but if it regular, one possibility if that the esp8266/esp32 is rebooting. In my very limited experience, this has happened for electrical reasons, that is, either faulty power supply, or some device that draws too much current from the esp32. Many relays will draw more than the 40 mA max if connected directly, without a relay driver circuit between. But the common relay modules will include this, so it depends what you have.

vitotai, 2021-01-12:
what @daghovland  said.

Nauticno, 2021-02-07:
Hi! I found soultion. It was using a to weak power. It is OK now!

closed 2021-02-07 by Nauticno