vitotai/BrewPiLess #307: Cannot calibrate temperature sensor

daghovland, 2021-01-22:
Hi! One of my temperature sensors (DS18B20) for some reason consistenly gives exactly 10 degrees C more than the others. When I try to enter "-10" in the calibrate field, the value is changed to "6". I have erased the EEPROM and made the setting from scratch, and still same issue. 

I am using v4.1, compiled from what is still the latest version of the master branch here in github. The hardware is TTGO T-display with ESP32

vitotai, 2021-01-22:
10 degree difference is weird. 
The calibration related code is derived from BrewPi. As far as I know, it is stored in 4.4 fixed point format. So, 1 bit for sign and 3 bits for integer part, which means maximum 7 degree difference.

I don't touch that part of code. You might need to find another sensor or other solution.

daghovland, 2021-01-22:
Ok. Thanks for the info. Guess I will have to scrap the sensor. Weird, though.

closed 2021-01-22 by daghovland