vitotai/BrewPiLess #310: Simple pressure sensor

neo-64, 2021-02-04:
Hi,

Can I use a simple pressure sensor like this one :
https://a.aliexpress.com/_vq9hhV

With esp8266 GPIO15 ? I need a resistor ?

Regards

vitotai, 2021-02-05:
Yes and No.
The pressure transducer is fine, but 300psi is way too high. The highest pressure tolerance of most container in beer brewing as far as I know is around 60 to 120psi, or lower. Too high maximum would usually result in low resolution.

You can't use GPIO15.

Read the document. 
https://github.com/vitotai/BrewPiLess/blob/master/doc/PressureMonitor.md

neo-64, 2021-02-05:
this sensor exist in 60psi-5VDC-0.5-4.5V
thanks for the link i'll use external ADS1115  as recomended, maybe one day i would like to automate spunding

closed 2021-02-05 by neo-64