vitotai/BrewPiLess #313: Heating for only a couple of minutes

whywouldi, 2021-02-13:
I have the three temp probes (Beer, fridge & room) and a pressure transmitter connected to the latest version (4.2). 

Beer is at 14 set temp is 19. "Waiting to heat" cycles for 5 minutes then it heats for approx 2 minutes before moving to "waiting to heat" again. I have changed the beer temp sensor out incase this was faulty but still the same issue.

any help would be great.

whywouldi, 2021-02-13:
So......I have replicated this on another brewpiless setup using 4.1. 

I have completed the following fault finding checking at each stage for correct operation:

Changed all temp transmitter a number of time.
disabled pressure
moved software back to 4.0r1
wiped eprom and reinstalled 4.1

I am still having the same issue with heating only going on for approx 2 minute (this is not a fix time it does vary).

I have noted:
The temp readings in devices are somtimes not there.
Brewfather is not logging despite playing with the setting at each stage.

I am now stumped!

whywouldi, 2021-02-13:
Managed to sort it.

Noticed that the temperature reading were indeed dropping out randomly. This was causing the heating circuit to suspend/reset as it was not sure what the temperature references were. It finally turned out to the PSU! I had a 5v supply and was using the D1 mini 3.3v for the temp transmitters. All voltages were correct so this was the last thing I changed. Now I powering from the micro USB on the D1 mini and everything is working ok.

Seems very strange for this to be the issue but maybe someone with better understanding of the board could explain.

Hope this helps someone as I have spent all day fault finding this one 🤦‍♀️

daghovland, 2021-02-13:
Your issue and solution are not possible to reproduce without more details. Are you using the DS18B20 temp sensors? Have you connected the pull-up correctly? Both 3,3 and 5 volt work fine with DS18B20, both on spec and in my experience.

whywouldi, 2021-02-14:
I am using DS18B20’s with the 4k7 ohms resistor connected between the 3v3 & the signal cable. I had a system thats been operation for a year so am happy with the operation and setup. The reason for change is to incorporate the pressure reading and spunding control (although I am also struggling to get D0 to recognize the relay module at present for the control but the pressure is working great). Anyway the PSU I was using when the issue was present connected the D1 mini ESP8266, the ADS1115 and the pressure transmitter. The 3v3 output from the D1 mini was used to power the temp transmitters. All voltages were checked throughout (D1 mini 3v3 reading was 3.333v). I also monitored the voltage while witnessing the data drops. At no point did I see a spike or drop in voltage (although using a digital process volt meter there is the potential to miss an extremely short blip).

As soon as I replaced the stand alone PSU for the micro USB supply direct into the D1 mini everything was ok. My new setup was two systems (for my two fridges) so I was able to check the problem on two identical setups and the issue replicated on both. It was the last thing I changed as the voltages were spot on so no cause for concern.

Hope this additional detail helps. If there are others out there powering the setup using a single 5v PSU I wonder what the issue is with the PSU. I could put it on a data recorder at work to see if there are any spikes etc but to be honest now I have it working I will go with this setup.

Don’t want to go off on a tangent but if you have any ideas about the D0 relay output not being recognized that would be very helpful.

Cheers,

daghovland, 2021-02-15:
_Don’t want to go off on a tangent but if you have any ideas about the D0 relay output not being recognized that would be very helpful._

Cannot know for sure without more information about the module and your circuit, but most cheap relay modules are 5 volt, but many of them also work at 3.3 volt. If it is a module with driver circuit and power in (and not just bare relay), the power in must often be the same voltage as the signal from the D1. So if you are powering the module with 5 volt that might be the reason. If the D1 can handle powering the relay, try using the D1 3V pin. Also test the relay module by connecting the signal pin to 3V on the D1. If that still does not work, you need separate 3 volt supply, a relay driver circuit (basically just a resistor and a transistor), or a new relay,

whywouldi, 2021-02-18:
Sorry for the delay in reporting back.
So the relay is operating correctly. I’ve connected it to D0 as per documentation for D1 mini (only three possible connections D7 I have as cooling, D5 for comber heater and......D0 attempting to use as spunding solenoid valve). When I power the module up the D0 relay click on-off-on (staying on). When I go to devices to detect and set it up as spunding it is not recognized…
Maybe the D1 mini can’t use D0 so I’m thinking of getting a D1 mini pro for additional outputs to do what I want.
Cheers,

vitotai, 2021-03-01:
D0 is used for Buzzer by default.
If you need more pins, ESP32 would be a better choice. All ESP8266s are the same, be it a D1 mini, D1 mini pro, or Nodemcu.

whywouldi, 2021-03-01:
Ok thank you.
I’ve now purchase a ESPwroom32 so will give that a go 👍🏻.
Appreciate your help.