vitotai/BrewPiLess #317: BPL doesn't create WiFi network

Stolowma, 2021-03-08:
I flashed a D1 Mini pro with the ModeMCU flasher (16m flash size) and BrewPiLess.v4.1r4.d1pro.newui.16m14m.bin. It appears to have flashed OK but the brewpiless network never shows up on my wifi.

daghovland, 2021-03-09:
I have  used brewpiless on a D1 mini (not pro). I cannot recall it flashing the onbaord led, and not on any other chips I have used. Is it supposed to do that? Why are you flashing it with NodeMCU, and not directly with USB? Have you tried flashing some simple debug programs, for example one that flashes more frequently? (Just so you are completely sure the flashing process works) Does the flashing process give any errors or warnings? Have you tried reading the serial output from the D1 mini via usb? (Not sure if your build has that turned on, but worth trying). Has the chip been used with other BPL installations earlier? The wifi settings are not deleted when flashing BPL, so any wifi settings from previous installations are still there. I see some D1 mini pros are sold without built-in antenna. Could it be you need to connect an external antenna? Is your power supply stable? (The flashing could potentially mean repeated booting)

Stolowma, 2021-03-09:
Thanks for your answer
I've come to the conclusion that the module is bad. I flashed another and it works fine and the original one won't flash at all now. I did use the more generic bin instead of the one that apparently is supposed to be for the Pro. I assume the difference is it takes advantage of the extra memory the Pro has.  I think I'll go with the if it works don't fix it philosophy and leave it as is.

closed 2021-03-09 by Stolowma