vitotai/BrewPiLess #319: can't see D0

neo-64, 2021-03-22:
Hi all,

I try to mount a sunding, so i would like to configure D0 for solenoid.

In the device list i can't see D0, i had try init/erase EPROOM but this action don't solve my issue.

I have made OTA upgrade from 3.7 to 4.1 maybe i'll need to re-install from usb ? I use : BrewPiLess.v4.1r4.4m2m.bin for a wemos D1 mini, it's the good one ?

Soldering seems to be good do you think i need to remake the soldering ?

regards

neo-64, 2021-03-23:
ok finally i have found a workarround to connect my solenoid to brewpiless 👍 

1)used this setup : https://github.com/vitotai/BrewPiLess/blob/master/doc/HardwareExample1.md
2)Mount presure sensor with external ADS1115
3)Use pin D8 for the relay of the solenoid

**Issue with this setup : my wemos D1 mini don't boot if D8 pin is pluged, the workarround i have found is to boot with D8 disconnected and connect D8 to the relay after boot.**

This setup seems to work fine, i just need to find a clean solution for booting with D8 connected, for now i'll just add a little manual switch for booting

(sorry for my bad english)

vitotai, 2021-03-24:
D0 has never been available. It is supposed to be used for Buzzer, which is in fact not used by BPL.
Please read this page and linked page for further information about PINs of ESP8266.
https://github.com/vitotai/BrewPiLess/blob/master/doc/HardwareGeneral.md

On my ESP8266, D8 works fine to control a relay. D8 must be low to enable normal boot.(see the document above). I have checked some schematics and found that D8 is pulled low by a resistor. Maybe your ESP8266 board misses that pull down resistor. 
Anyway, I don't recommend you keep using it. The system will not boot-up in case of power shortage. Or, if for some reason  it reset, the system is locked. ESP32 might be a better solution if you need more PINs.
Usually, people don't really need cooling and heating at the same time. You might spare one.

Although my BPL has once run for a few months without rebooting,  sometimes it resets a few times during one fermentation. Maybe it's because my crapy setup (the simple setup in the document you mentioned), or maybe it has something to do with the underlying WiFi related tasks. Making sure your system will reboot successfully is very important.

neo-64, 2021-03-24:
Thanks for your reply, there's something not very clear in the documentation : 

https://github.com/vitotai/BrewPiLess/blob/master/doc/AutoSpunding.md

"Auto capping is enabled by assigning an actuator, a pin, in "Device Setup". Currently, only three pins can be used on ESP8266. They are cooler, heater, and door pins, default to D7, D5, and D0(after v2.7)."

Using ESP32 seems to be the safe, but i need to rework all my setup, i can't do that quickly. Is there any way to : 
-Normal boot with D8 on relay (i haven't seen clear schema for that)
-Hard code D0 for capping relay (maybe modify config.h and make a custom build ?)
-Using  Expansion Board I/O Expander I2C-Bus, do you think it could work without code modification ?

Thanks vitotai for this amazing job with BPL i have the best geek fridge fermentor EVER :)

vitotai, 2021-03-24:
That's out dated. I made it to D8. Sorry about that.

The correct answer is "coolingPin", "heatingPin", and "doorPin", although they are assigned in setup page. It's legacy code from BrewPi.

The schema I red is ESP12, usually the base of NodeMCU or D1 mini. GPIO15 is pulled low.
https://docs.ai-thinker.com/_media/esp8266/docs/esp-12f_product_specification_en.pdf

You might change the doorPin in config.h to rebuild. It would appear in setup page. The reason I don't use D0 is that it outputs random signal during booting and it's also a special PIN.

You might need to your relay module. It seems to pull input pin to high.

Scales82, 2021-04-02:
I have my BPL boards set up for D0 for spunding, I use thorax's bin to make it work. Should I be concerned by your above comments @vitotai ? Or can i safely continue using D0? I too took your outdated statement as advice when I built my custom PCB's (made several too)

vitotai, 2021-04-04:
The problem with D0 is that it outputs random signals when booting. (The signals might mean something to the MCU developers, but it's noise to us.) Therefore,
1. It happen only at booting. If your system doesn't reboot frequently, it should be fine.
2. If the signal triggers the relay connected, your device might run and stop in a very short periods for multiple times. The frequency is so hight that I doubt the relay may not response. Even if the relay responses, the controlled  device might not start and stop as the relay on and off. Even so, it's bad to a fridge, and it might not hurt a solenoid.

So, you should be fine.

neo-64, 2021-04-06:
ok,
I have found à workarround : i have add a diode 1N4007 between GPIO15 and relay board command. It's work like a charm, just when esp boot there's a quick on/off on the relays for this pin. I'll set up GPIO 15 on heating because i don't want solenoid open on boot and it will be great for me !