vitotai/BrewPiLess #322: Cannot see fourth relay - misidentifying as sensor

delight-f, 2021-04-03:
Thanks so much Vitotai for the great built. I'm using a four-relay board with a built-in esp-12F.

However,under my devices screen it only shows three relays and one sensor device. No sensors are connected. Any ideas why it may be misidentifying a relay as a sensor? [image]

delight-f, 2021-04-03:
I can't work out how to change the pins in the Config.h file.
It has defined NodeMCU_PIN but on a bare ESP-12F board there are no pin markings!

vitotai, 2021-04-04:
Forth control pin never exists on ESP8266.
https://github.com/vitotai/BrewPiLess/blob/master/doc/HardwareGeneral.md

delight-f, 2021-04-05:
> 
> 
> Forth control pin never exists on ESP8266.
> https://github.com/vitotai/BrewPiLess/blob/master/doc/HardwareGeneral.md

I'm confused, because it is identifying a controller pin attached to GPIO 15 - according to my schematic, this IO is connected to ground!

I can therefore only see two control pins connected to relays where I should be able to see three  - please see screenshot.

vitotai, 2021-04-10:
You should read ALL the code before changing part of it.
Also, read the linked pages in the document provided. 
Last, my D8 works for me.

closed 2021-04-10 by delight-f