vitotai/BrewPiLess #324: MQTT remote control issue with Path for Spunding control

neo-64, 2021-04-06:
Hi,

When i enter "path for spunding control" after "send" textbox is blank. I think there's a bug here... any workarround ?

regards

vitotai, 2021-04-10:
I am not sure I understand you. Maybe a picture?

neo-64, 2021-04-12:
sorry my english is really poor...

if i set up this field : 
[image]
then i click send button and when i refresh the page this field is blank and mqtt remote control topic don't work

Is there à REST API or something like that to send this kind of command ?
https://github.com/vitotai/BrewPiLess/blob/master/doc/JsonCommand.md this doc is the complete list of command or there's more commands ?

Regards