vitotai/BrewPiLess #325: ISpindel issue

cmckay22, 2021-04-17:
I have been testing v4.2 just to see the functionality and seem to have found an issue. I realize it is still in early stages and not using it bit thought is would mention it. Seems every time my Ispindel sent an update the brewpiless would reboot. My ispindel is running the latest version and is working fine with the current Brewpiless version. Have recreated on both Wemos d1 and pro version.

vitotai, 2021-05-14:
It works for me. Maybe you could provide more information, like your settings. version of iSpindel, and etc.

cmckay22, 2021-05-14:
I am running it on a Wemos d1 pro board using the prebuilt image you compiled. As for the ISpindle it is running version 6.5.1 (on a wemos d1 pro) from universam1 repository again the pre-compiled version. I was just setting up the brewpiless box and had configured the basic setup for ISpindel with the it set as beer temp. Attached are a few screen shots of the setup and the /tempcmd aswell as the graph page before, during and after the failure.  As mentioned before this is not a pressing issue and I appreciate all the work you put in. Was only testing the new version because the rotating LCD would be a nice to have. Let me know if you need anymore info or want me to test something. I have a basic setup build for testing before adding it to my fermenters
[image]
[image]
[image]
[image]

vitotai, 2021-05-15:
The invalid gravity causes the crashes. The code was not handling that well.
Try new pushed one.