vitotai/BrewPiLess #326: I2C LCD blank screen

Hugo-Pereira, 2021-04-21:
Hi. I got this I2C LCD https://www.aliexpress.com/item/32991449983.html?spm=a2g0s.9042311.0.0.682a4c4djWkZ2n
I connected everything as per the image provided (https://github.com/vitotai/BrewPiLess/blob/master/doc/image/BPL_simple.jpg), with the exception that I am powering the nodeMCU with USB, and not using any buttons (don't have them).

I was able to read the temperature from the 2 probes, and am also able to control the relay. However, the LCD is blank. It lights up, but doesn't show anything.

I tried to upload esp9266-dev but, although it compiles fine, I get an error when uploading (missing files). I can upload "newui" though. I am not sure if "newui" is the whole firmware? Or should I flash multiple modules like newui + rsp9266-dev? The way I got it working the first time was to just upload all. Some failed, but the device works with the exception of the blank I2C LCD.

I also tried flashing https://github.com/vitotai/BrewPiLess/blob/master/bins/BrewPiLess.v4.1r4.4m2m.bin, and got the same results. Everything works, except for the I2C LCD.

The detected device list is also weird. It detects a lot of devices even with nothing connected.
This is what I get with everything connected:
[image]

I am powering the LCD with 3v3 pin below D4 (as per the image). Tried using the Vin pin, but with same results.

PS: I am a complete novice regarding Arduino and electronics :)

stefschin, 2021-04-21:
Hi Hugo, this LCD doesn't work with 3.3V, just with 5V. Good luck

Hugo-Pereira, 2021-04-21:
I powered it with the vin pin also, and it didn't work. Shouldn't it have worked?

stefschin, 2021-04-21:
Yes, Did you check for contrast? it's a blue potentiometer on the back of I2C LCD

Hugo-Pereira, 2021-04-21:
Don't really know what that means. This is the back of the device 
[image]

closed 2021-04-21 by Hugo-Pereira