vitotai/BrewPiLess #327: Help, abnormal operation in esp32

jaimeagaviria, 2021-04-26:
Hi, I have started brewpiless on an ESP32, the system works for a few hours, but then the display shows strange characters and stops transmitting data to thingspeak.

My configuration is very basic, I have a 5v power supply connected to the vin pin of the nodemcu esp32, the display is also powered from the power supply and not directly from the arduino.

[image]

[image]

Could you please help me, thank you very much.

sdjngyq, 2021-05-13:
Please send the picture of the whole circuit to me. I suspect it is caused by the loose contact of your wiring circuit.

jaimeagaviria, 2021-05-13:
Hello, after some time I have solved this error as follows:

* I have put a 4.7 pull up resistor in the middle of the SDA and 5V pin of the LCD display.

* I have put a 4.7 pull up resistor, in the middle of the SCL pin and 5V of the LCD display.

* I have put a ceramic capacitor in the middle of the 5V and GND pins that connect to the LCD display.

Apparently the problem was caused by electrical noise.

closed 2021-05-13 by jaimeagaviria

sdjngyq, 2021-05-13:
OK，congratulations.

palmtreefrb, 2021-05-25:
> Hello, after some time I have solved this error as follows:
>
> * I have put a 4.7 pull up resistor in the middle of the SDA and 5V pin of the LCD display.
> * I have put a 4.7 pull up resistor, in the [...]

What capacitor capacitance did you use?

jaimeagaviria, 2021-05-25:
Hi palmtreefrb.

My underlying problem occurs when the power cables of the lcd display are very close to the power supply of the arduino, I use a 5V 3A supply.

I used a 100uf ceramic capacitor

Kind regards