vitotai/BrewPiLess #330: consult, ESP32 added tftlcd  after the system reset irregularly,

sdjngyq, 2021-05-13:
ESP32 Brewpiless adds a 3.5-inch TFT display screen, and only adds several external data receiving and display of ISPINDEL. The operation is basically normal, but the system will be restarted irregularly.
Sometimes it works steadily for 12 hours, sometimes it restarts within a few hours, sometimes less.
 The power supply is 5v1.5A, which basically eliminates the power supply problem.No modification has been made to the watchdog part in the program. Could you please tell me what other reasons may cause the ESP32 to restart?

vitotai, 2021-05-14:
defective ESP32 board. One of them.

sdjngyq, 2021-05-20:
Thank you for your reply，Several reasons were analyzed:  Basic elimination of voltage and current problems .
16 MB of Flash was used.
partition.csv   ,Maybe there's a distribution problem here  ,Please confirm if this is the problem?
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x600000,
app1,     app,  ota_1,   0x610000,0x600000,
spiffs,   data, spiffs,  0xc10000,0x200000,

vitotai, 2021-05-21:
Some people had issues with 16MB ESP8266. The reason is still unknown to me. 
You will need to try your hardware by yourself.

sdjngyq, 2021-05-21:
Thank you for your reply.  I'm using a 16M  esp32.

sdjngyq, 2021-06-13:
> Some people had issues with 16MB ESP8266. The reason is still unknown to me.
> You will need to try your hardware by yourself.
Hi ， vitotai   
This is part of the running log table, restart at this time in the red circle, and analyze what caused this

[image]