vitotai/BrewPiLess #331: Sending data to thingspeak stops after some time using ESP32

jaimeagaviria, 2021-05-17:
Hello, I have 2 fermenters controlled by brewpiless, the first one works with an esp8266, it works very well and sends me data every minute through thingspeak, I have installed a second controller but using an ESP32, it also sends data to thingspeak every minute, however after some time it stops transmitting, both controllers are very close to the wifi antenna.

For the ESP32 controller to send data again I have to turn it off and on again

Has the same thing happened to someone?

Is it possible that the version for ESP32 is less stable than the version for ESP8266?

Thank you,

Hwerow, 2021-05-19:
Interesting, I have had a similar issue with ESP32 stopping sending data to Brewfather (set for 15 minute intervals). I thought that it may have been a power spike or something as this seems to happen randomly, as it might work for 2 days then stop. I have to reset the ESP32 to get it to send again. BPL still keeps logging and backfills the data. 
This has happened over several brews with version 4.1 esp32-oled
Vito is there some difference in the way the ESP32 works compared to the 8266?

jaimeagaviria, 2021-05-20:
Hello, another thing that I have observed, in the ESP8266 when there is an internet disconnection or the Wi-Fi router is disconnected, the esp8266 reconnects automatically when restoring internet access and begins data transmission, however the EPS32 version does not it works the same.

Kind regards

vitotai, 2021-05-20:
The underlying libraries and hardware differ.
Even on ESP8266, some versions of framework seem to recover network while some seems not.

My ESP32 build is still not yet finished, while my ESP8266 with latest version is now running the second brew. 
I will test ESP32 later to find out what happen.

vitotai, 2021-05-22:
updated:
I did some simple test.
1. Using sample code which hard coded my SSID/password, and my ESP32 could NOT connect.
2. Using the WiFiSetup I used in BPL to handle reconnect, and the ESP32 connected and lost connection after a while, then connected and lost again. 

Maybe I should remove ESP32 build and declare ESP32 is not supported?

Hwerow, 2021-05-22:
Well my ESP32 with version 4.1 connects OK and maintains its connection (currently over 6 hours) on the current batch -  it renders the Mode Beer Temp etc OK and it controls temperature hot and cold OK. The problem seems to be that the graph is not rendered on iOS or Firefox or Chrome for some reason which suggests some other problem. I think not supported is a bit too harsh? Perhaps under development use with caution?

vitotai, 2021-05-23:
Good news, or bad one to some of you, including me.

I tried my other ESP32, and it works very good.

jaimeagaviria, 2021-05-23:
Hi Vito, for how many hours have you done the test? I have 3 Brewpiless implemented, the esp8266 is like a rocka, very stable sending data, my ESP32s are irregular, when they turn on they behave well, when many hours pass the behavior happens, important the temperature control continues to work well even without transmitting data .

jaimeagaviria, 2021-05-24:
Hi Vito, in order to solve this without having to go back to esp8266, how can I see the debug information of esp32? Can the local log show me information about what is happening in the BPL system?

vitotai, 2021-05-24:
At least 12 hours, I had tested. Maybe I had it run over weekend when porting BPL to ESP32.
On the other hand, the "bad" ESP32 might lost network "forever" and never get back. Really unstable.

You canbuild esp32-dev, which is the setting I used for development to see the logs. However, in this case, we can only know network lost, and BPL trying to recover, and etc. The exact reasons of lost are unknown. It's deep in the core that we don't have access to.