vitotai/BrewPiLess #334: Strange temp probe behavior at higher temps

Scales82, 2021-05-31:
Hi,
Im seeing some strange temp probe tracking at higher temps (around 35dec C) As seen in the graph, the fridge temp all of a sudden begins to follow roughly the same temp as the beer temp. (see blue line following green line below) Going into the device page and clicking "apply" on the temp probe seems to rectify it and wakes it back up to what the temperature in the fridge actually is....
Any ideas?? seems to only be at a higher temp range, im using KVEIK yeast and noticed it a couple times now on these brews.

[image]

glenosAU, 2021-06-24:
I have similar issues with my set up although they appeared after I built my controller into a box. I think it is either feedback from the eBay 5V power supply or mixed signals from the temp probes.

How it behaves is the room temp drops out to no result coinciding with the fridge temp changing to match the beer temp and then going back to the actual chamber temp. This causes problems with the control logic because it sees the temp as unstable and stops heating/cooling.

I plugged the fridge into the wall socket today to get it to cold crash, worked too well while I was at work down to -3.5°C.
[image]

Scales82, 2021-07-27:
Yeah that seems a little different to mine. I had that happen with an earlier iteration board which only had 3v3 supply to the probes. 
Update on mine, it happens at all temps now, but only when the fridge temp is close to the beer temp. Happened again today (and a few times since OP)

[image]

you can see where it starts to track the same as beer temp. I have replaced the probe with a new one, so i can rule that out. I am thinking about removing the room temp probe to see if I just have overloaded the D1

vitotai, 2021-08-22:
When fridge sensor is "disconnected", the temperature of fridge reading will come from "beer sensor". It's designed for glycol temperature control.

I have no explanation for lost of sensors. In fact, it happened to me a few times. Well, not that rare before I rewire my controller, but once a few weeks ago.
I uploaded an version built with debugging options on, and my fridge sensor was gone. After updating with another version, the fridge came back alive. The upgrading was done via OTA, and nothing except WiFi signal touches the controller and the fridge. 

It's difficult for me to find out the real problem, but I guess it has something about loading.

Scales82, 2021-08-28:
I have replaced the 4.7k sensor signal 3v3 pullup resistor to a 1k resistor as per a post elsewhere here on the git. sensors are 2m long and 5v powered. I am monitoring for last couple days but so far so good. Thanks for the explanation vitotai. I'll report here if the situation deteriorates.

glenosAU, 2021-08-30:
Based on @vitotai answer. My fridge probe is being list intermittently so the temperature tracks to the beer temp. But is coming back to life before it gets there and tracks back to where it should be. Seems to make sense, I also deep loss of room temp sensor.

Scales82, 2021-09-06:
so far so good on my 1k resistor change. Think it may have fixed the issue. I would have had a problem by now.