vitotai/BrewPiLess #336: Not working properly

jsbjx, 2021-06-23:
Test multiple ES8266, ESP32 firmware. No one can run normally for more than 20 hours. Troubleshoot hardware and power supply issues.

jsbjx, 2021-06-23:
https://user-images.githubusercontent.com/32890201/123028283-c33cb400-d411-11eb-9dbc-89f0b345607f.mp4

jsbjx, 2021-06-23:
Turn on, no signal is received. 19 hours jump to 11 hours

vitotai, 2021-06-24:
I don't understand your issue.
I have one that survives through 2 brews, or 2 months until I re-flash an image with debug functionality.

jsbjx, 2021-06-24:
Can you see if the above time will exceed 20 hours

vitotai, 2021-06-24:
Never.
The reason is that uint16_t is used to carry time duration in display.
65,536/(60*60) = 18.20 hours.
You can check the code in DisplayLcd.cpp

I don't think it maters and don't want to change that part of code, which is derived from original BrewPi.

jsbjx, 2021-06-25:
> 绝不。
> 原因是 uint16_t 用于在显示中携带持续时间。
> 65,536/(60*60) = 18.20 小时。
> 您可以查看 DisplayLcd.cpp 中的代码
> 
> 我认为它不重要，也不想更改源自原始 BrewPi 的那部分代码。

Thank you for your reply, it is my misunderstanding of the procedure.