vitotai/BrewPiLess #338: mDNS not working on v4.1

allthepies, 2021-06-24:
Just upgraded to v4.1 and mDNS is no longer working.   brewpiless.local no longer resolves and "dns-sd -B" does not find brewpiless.

Tried putting it into just station mode but still no joy.

lekrom, 2021-07-02:
I get the same issue.

pedro-jorge-pereira, 2021-07-02:
I also have the same issue.

vitotai, 2021-07-03:
My working setup has the same issue while my testing platform works just fine.
I guess it's a timing issue.

allthepies, 2021-07-03:
I tried running BPL on a D1 Mini Pro and mDNS worked on that, rebuilt the image for a basic D1 Mini and no mDNS, so something odd is happening.

vitotai, 2021-07-05:
I've done some experiments, the result of which makes me believe its something related to timing. The real reason is not clear before tracing and understanding all MDNS related code, maybe including LWIP.

A simple solution that helps but dose not solve the issue.
Adding the following code before starting WiFi(around line 1973 in BrewPiLess.cpp)
`
WiFi.setAutoConnect(true);
//1. Start WiFi (add before this line)
`

optional, add some code after initialization of WiFi ( a few lines after the starting code)
`
DBG_PRINTF("WiFi Done!\n"); // add the following code after this line
	int wait=5;
	while(wait>0 && WiFi.status() != WL_CONNECTED){
		delay(1000);
		wait--;
	}
`

Short explanation:
I found that if WiFi is not connected before MDNS.begin() is run, MDNS doesn't work. WiFi.setAutoConnect(true) requests WiFi connection at start-up instead of WiFi.begin(). The second part of code just waits for connection of WiFi.

BTW,
MDNS library changes a lot in latest framework. The problem might not be present in the latest framework. However, migration of frameworks requests some works and time for verification.

allthepies, 2021-07-05:
Thanks vito, I'll add the setAutoConnect call to my test version and give it a go.

allthepies, 2021-07-05:
Looking at the code, line 1973 doesn't look correct, do you mean 1873 ?
Like this...

```
	theSettings.load();

	SystemConfiguration *syscfg=theSettings.systemConfiguration();
	
	display.setAutoOffPeriod(syscfg->backlite);
	Wifi.setAutoConnect(true);
	#ifdef ENABLE_LOGGING
//  	dataLogger.loadConfig();
  	#endif

	//1. Start WiFi
	DBG_PRINTF("Starting WiFi...\n");
	WiFiMode wifiMode= (WiFiMode) syscfg->wifiMode;

```

vitotai, 2021-07-05:
That should work. 
However, it might take effect after reboot.(wherever you put the setAutoConnect()).

allthepies, 2021-07-05:
Yes, it worked with the additional delay.  I also modified the MDNS.begin to include an IP address (varies depending upon which AP mode is selected).  As apparently the IP address bound to by the MDNS implementation has changed depending upon the library level.

https://github.com/esp8266/Arduino/issues/7210#issuecomment-612631296

```
	DBG_PRINTF("WiFi Done!\n");

	int wait = 5;
	while (wait > 0 && WiFi.status() != WL_CONNECTED)
	{
		delay(1000);
		wait--;
	}

	// get time
	initTime(WiFiSetup.isApMode());

	if (!MDNS.begin(syscfg->hostnetworkname, WiFiSetup.isApMode()) ? WiFi.softAPIP() : WiFi.localIP())
	{
		DBG_PRINTF("Error setting mDNS responder\n");
	}
	else
	{
		MDNS.addService("http", "tcp", 80);
	}
```

vitotai, 2021-07-06:
The latest version of MDNS in 3.0 framework will ignore the IP parameter in MDNS.begin().

I've tried to call MDNS.begin() when WiFi is connected, but it doesn't work for an unknown reason. Since the library has changed a lot and this issue might not happen in latest framework, this temporary solution will be adopted.