vitotai/BrewPiLess #339: OLED firmware restarts continuously

jsbjx, 2021-07-01:
BrewPiLess.v4.2a4.4m2m.bin
BrewPiLess.v4.2a4.oled.newui.4m2m.bin

There are two kinds of firmware. Same motherboard. The LCD version is normal. The OLED version is constantly rebooting. What is the reason?

[image]
[image]

https://user-images.githubusercontent.com/32890201/124131999-f1299480-dab2-11eb-9204-574998b478ed.mp4

vitotai, 2021-07-03:
The cycling LCD display is under verifying and development. I have tested it on OLED while all my OLED LCDs are broken.

jsbjx, 2021-07-05:
Hello, I can provide the hardware you need for your experiment for free, including the experiment board of ESP8266 and ESP32, and also provide the display screen. How can I mail it to you?

vitotai, 2021-07-06:
Thank you, but what I need is time.
I am hunting an issue of frozen display right now. It hasn't happened since I adding some debugging code a week ago.
The issue of your LCD is strange, I can't reproduce it, either.

vitotai, 2021-07-06:
The OLED issue can be reproduced without hooking OLED. 
It seems that updating LCD takes too long to be done in HTTP session. I've made modification. Please try a5.

jsbjx, 2021-07-06:
a5 where? Please send a link, thank you!

vitotai, 2021-07-07:
https://github.com/vitotai/BrewPiLess/tree/4.2pre

Building and pushing take some time to finish.

jsbjx, 2021-07-07:
Tested A5 8266 Bin and no restart phenomenon

jsbjx, 2021-07-07:
[image]
[image]
[image]
[image]
These are the four display states of OLED. The bottom row is scrolled. It is recommended to fix the time display on the first page and the IP display on the second page. So there are only 4 lines on the second page, and one line of voltage display is added.

jsbjx, 2021-07-07:
[image]
[image]
There is only one Beer temperature sensor in the hardware setting, set Beer Set 20. Fridge temp:30 Fridge Set:10 These two are automatically generated?

prosenbaum91, 2021-11-03:
> [image] [image] There is only one Beer temperature sensor in the hardware setting, set Beer Set 20. Fridge temp:30 Fridge Set:10 These two are automatically generated?

Can you send me a bin with version 4.2 for oled and wemos d1 mini? I can't comple it.

jsbjx, 2021-11-03:
Don't use this software anymore. many problems. I have given up.

jsbjx, 2021-11-03:
> > [image] [image] 硬件设置里只有一个啤酒温度传感器，设置Beer Set 20。Fridge temp:30 Fridge Set:10 这两个是自动生成的？
> 
> 你能给我寄一个 oled 和 wemos d1 mini 的 4.2 版本的 bin 吗？我无法完成它。

Don't use this software anymore. many problems. I have given up.

prosenbaum91, 2021-11-03:
I have a problem with version without oled too. What version is stable? I have countinousely reboot.