vitotai/BrewPiLess #341: Log version mismatched while logging in BPL 4.1r4

lorenjanwilson, 2021-07-17:
After logging for about a day or more, I get the "Log version mismatched!" error and need to stop and restart logging to be able to see the graph again.

Running BrewPiLess.v4.1r4.4m2m.bin on an esp8266 D1 mini.

I currently have it in this state, so let me know if I can collect any logs for you.

Seems possibly related to [this issue](https://github.com/vitotai/BrewPiLess/issues/276), same symptoms but different BPL version. I can delete this and post there if you prefer.

lorenjanwilson, 2021-07-17:
By the way, tried clearing all browser data, force-reload, and tried a new fresh browser as well as a private window. Everything gets the same error.

lorenjanwilson, 2021-07-17:
Here is what happens when I end a "log version mismatched!" log and download it. (I gzipped it so it would upload, you can ungzip it)

[Basement.gz](https://github.com/vitotai/BrewPiLess/files/6834854/Basement.gz)

vitotai, 2021-07-27:
I checked the file and found that somehow the log has TWO flags of "log beginning" flags, which is weird. Have no clue.

lorenjanwilson, 2021-07-28:
Thank you for taking a look! I downgraded this same device to BrewPiLess.v3.6.4m2m.bin and I can log for several days in a row... no "log version mismatched" error with v3.6.