vitotai/BrewPiLess #342: Build trouble xtensa/hal.h

bsaurusrex, 2021-07-23:
Windows platformIO build. Stuck on this error.

In file included from C:/Users/brian/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/FreeRTOSConfig.h:103:0,
                 from C:/Users/brian/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/esp32/include/freertos/include/freertos/FreeRTOSConfig.h:75,
                 from C:/Users/brian/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/esp32/include/freertos/include/freertos/FreeRTOS.h:58,    
                 from C:/Users/brian/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/esp32/include/esp_event/include/esp_event.h:20,
                 from C:/Users/brian/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/libraries/WiFi/src/WiFiGeneric.h:27,
                 from C:/Users/brian/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/libraries/WiFi/src/WiFiSTA.h:28,
                 from C:/Users/brian/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/libraries/WiFi/src/WiFi.h:32,
                 from src/OTAUpdate.h:4,
                 from src/OTAUpdate.cpp:1:
C:/Users/brian/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/xtensa_config.h:40:24: fatal error: xtensa/hal.h: No such file or directory
compilation terminated.
Compiling .pio\build\d32_pro_tft\src\bridge_lcd.cpp.o
Compiling .pio\build\d32_pro_tft\src\http_server.cpp.o
Compiling .pio\build\d32_pro_tft\src\jsonconfig.cpp.o
Compiling .pio\build\d32_pro_tft\src\main.cpp.o
Compiling .pio\build\d32_pro_tft\src\sendData.cpp.o
Compiling .pio\build\d32_pro_tft\src\serialhandler.cpp.o
Compiling .pio\build\d32_pro_tft\src\tilt\tiltHydrometer.cpp.o
In file included from C:/Users/brian/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/FreeRTOSConfig.h:103:0,
                 from C:/Users/brian/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/esp32/include/freertos/include/freertos/FreeRTOSConfig.h:75,
                 from C:/Users/brian/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/esp32/include/freertos/include/freertos/FreeRTOS.h:58,    
                 from C:/Users/brian/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/cores/esp32/Arduino.h:33,
                 from .pio/libdeps/d32_pro_tft/ArduinoJson/src/ArduinoJson/Configuration.hpp:138,
                 from .pio/libdeps/d32_pro_tft/ArduinoJson/src/ArduinoJson.hpp:7,
                 from .pio/libdeps/d32_pro_tft/ArduinoJson/src/ArduinoJson.h:9,
                 from src/tilt/tiltHydrometer.h:5,
                 from src/tilt/tiltScanner.h:8,
                 from src/bridge_lcd.h:4,
                 from src/bridge_lcd.cpp:1:
C:/Users/brian/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/xtensa_config.h:40:24: fatal error: xtensa/hal.h: No such file or directory
compilation terminated.
Compiling .pio\build\d32_pro_tft\src\tilt\tiltScanner.cpp.o
Compiling .pio\build\d32_pro_tft\src\uptime.cpp.o
Compiling .pio\build\d32_pro_tft\src\version.cpp.o
*** [.pio\build\d32_pro_tft\src\OTAUpdate.cpp.o] Error 1
Compiling .pio\build\d32_pro_tft\src\watchButtons.cpp.o
*** [.pio\build\d32_pro_tft\src\bridge_lcd.cpp.o] Error 1
In file included from C:/Users/brian/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/FreeRTOSConfig.h:103:0,
                 from C:/Users/brian/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/esp32/include/freertos/include/freertos/FreeRTOSConfig.h:75,
                 from C:/Users/brian/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/esp32/include/freertos/include/freertos/FreeRTOS.h:58,    
                 from C:/Users/brian/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/cores/esp32/Arduino.h:33,
                 from src/uptime.h:8,
                 from src/http_server.h:4,
                 from src/http_server.cpp:2:
C:/Users/brian/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/xtensa_config.h:40:24: fatal error: xtensa/hal.h: No such file or directory
compilation terminated.
*** [.pio\build\d32_pro_tft\src\http_server.cpp.o] Error 1
In file included from C:/Users/brian/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/FreeRTOSConfig.h:103:0,
                 from C:/Users/brian/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/esp32/include/freertos/include/freertos/FreeRTOSConfig.h:75,
                 from C:/Users/brian/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/esp32/include/freertos/include/freertos/FreeRTOS.h:58,    
                 from C:/Users/brian/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/cores/esp32/Arduino.h:33,
                 from .pio/libdeps/d32_pro_tft/ArduinoLog/ArduinoLog.h:20,
                 from src/serialhandler.h:8,
                 from src/jsonconfig.h:4,
                 from src/jsonconfig.cpp:1:
C:/Users/brian/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/xtensa_config.h:40:24: fatal error: xtensa/hal.h: No such file or directory
compilation terminated.
*** [.pio\build\d32_pro_tft\src\jsonconfig.cpp.o] Error 1
In file included from C:/Users/brian/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/FreeRTOSConfig.h:103:0,
                 from C:/Users/brian/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/esp32/include/freertos/include/freertos/FreeRTOSConfig.h:75,
                 from C:/Users/brian/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/esp32/include/freertos/include/freertos/FreeRTOS.h:58,    
                 from C:/Users/brian/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/cores/esp32/Arduino.h:33,
                 from .pio/libdeps/d32_pro_tft/ArduinoLog/ArduinoLog.h:20,
                 from src/serialhandler.h:8,
                 from src/sendData.h:4,
                 from src/sendData.cpp:5:
C:/Users/brian/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/xtensa_config.h:40:24: fatal error: xtensa/hal.h: No such file or directory
compilation terminated.
[... 66 lines cut ...]
                 from src/version.cpp:1:
C:/Users/brian/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/xtensa_config.h:40:24: fatal error: xtensa/hal.h: No such file or directory
compilation terminated.
*** [.pio\build\d32_pro_tft\src\version.cpp.o] Error 1
In file included from C:/Users/brian/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/FreeRTOSConfig.h:103:0,
                 from C:/Users/brian/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/esp32/include/freertos/include/freertos/FreeRTOSConfig.h:75,
                 from C:/Users/brian/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/esp32/include/freertos/include/freertos/FreeRTOS.h:58,    
[... cut ...]
C:/Users/brian/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/xtensa_config.h:40:24: fatal error: xtensa/hal.h: No such file or directory
compilation terminated.
*** [.pio\build\d32_pro_tft\src\watchButtons.cpp.o] Error 1
================================== [FAILED] Took 3.29 seconds ==================================
Environment    Status    Duration
-------------  --------  ------------
d32_pro_tft    FAILED    00:00:03.295

Thanks

VeronicaGupta, 2023-08-24:
any update, did this resolve for you?

bsaurusrex, 2023-08-24:
So long ago, I cant remember anymore.

closed 2023-08-24 by bsaurusrex