vitotai/BrewPiLess #344: Local LCD display read out corrupted but functional operation continues ok

whywouldi, 2021-08-21:
I've had a number of brewpiless setups using ESP8266 chips and on all of them I find after being on for a few days the display output starts to corrupt. The brewpiless operation is flawless but when checking the local display I have to power cycle the circuit to get the LCD to display correctly.

I have tried power cycling just the display but this does not resolve the issue (actually makes it worse).

Is there a way to internally re-initial the display every 24 hours or something so everytime the display is readable?

vitotai, 2021-08-22:
You should check your wiring, or try your best to avoid EMI.

The LCD is re-initialized every 10minutes, but it doesn't work before current 4.2pre because of typo.
Solution for older version:
Searching display.fresh() in BrewPiLess.cpp, and replace it by display.refresh().

whywouldi, 2021-09-04:
I've been playing with the brewpiless but there are definaetly no loose connections and the issue is still repeatable. Once the current brew is completed I will attempt wiping the Eprom and re-flashing to see if it lessens the effect. 

thanks for the reply.

vitotai, 2021-09-06:
Re-flashing won't fix this issue.

https://labprojectsbd.com/2021/07/03/how-to-solve-noise-garbage-on-hd44780-lcd-display/

You could you 4.2pre branch.

closed 2025-03-08 by whywouldi