vitotai/BrewPiLess #349: Incorrect temperature regulation in 4.1

peteng72, 2021-11-14:
When I set the Fridge constant temperature setpoint to 12 deg, the measured Fridge temperature regulates between 12.5-13.0 deg.
I`m running version 4.1.

petobena, 2021-11-14:
check your settings https://github.com/vitotai/BrewPiLess/wiki/JSON-command

peteng72, 2021-11-14:
14:56:39 ↑ c
14:56:39 ↓ C:{"tempFormat":"C","tempSetMin":-5.0,"tempSetMax": 30.0,"pidMax": 10.000,"Kp": 5.000,"Ki": 0.250,"Kd":-1.500,"iMaxErr": 0.500,"idleRangeH": 1.000,"idleRangeL":-1.000,"heatTargetH": 0.299,"heatTargetL":-0.199,"coolTargetH": 0.199,"coolTargetL":-0.299,"maxHeatTimeForEst":600,"maxCoolTimeForEst":1200,"minCoolTime":180,"minCoolIdleTime":300,"minHeatTime":180,"minHeatIdleTime":300,"deadTime":600,"fridgeFastFilt":1,"fridgeSlowFilt":4,"fridgeSlopeFilt":3,"beerFastFilt":3,"beerSlowFilt":4,"beerSlopeFilt":4,"lah":0,"hs":0}

peteng72, 2021-11-14:
Tried to change idleRangeH with the command:
j{”idleRangeH”:0.300}

But still get exactly the same response to the ”c” command.

peteng72, 2021-11-14:
Seems like my double quotes are different than those used in BPL. Copied a response message and solved the problem.

closed 2021-11-14 by peteng72