vitotai/BrewPiLess #351: Humidity isn't being published to mqtt broker

shingle01, 2021-11-24:
I've noticed humidity isn't being published to mqtt.  Documentation confirms humidity is not included in topics.  It would be nice if it could be

shingle01, 2022-01-10:
Thanks for the recent commit to 4.2pre - 8c8709272251a684b94e214c2fa5f58a30150b27 to enable mqtt publishing of chamber humidity.

Looking through the changelogs I only see edits to the generic HTTP logging.

Are you able to update MQTT.md with the label?  May I suggest fridgeHumidity and roomHumidity

vitotai, 2022-01-12:
As you wish.
Please try latest push.

shingle01, 2022-01-16:
I misspelled Humidity earlier and it looks like you may have copied those label names, sorry.

I am now receiving mqtt packets on that topic but the data is invalid:

2022-01-16 11:53:20 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on brewing/brewpiless/state: b'1'
2022-01-16 11:53:20 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on brewing/brewpiless/beerTemp: b'22.9'
2022-01-16 11:53:20 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on brewing/brewpiless/beerSet: b'21.0'
2022-01-16 11:53:20 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on brewing/brewpiless/fridgeTemp: b'23.4'
2022-01-16 11:53:20 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on brewing/brewpiless/fridgeSet: b'23.0'
2022-01-16 11:53:20 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on brewing/brewpiless/roomTemp: b'24.7'
2022-01-16 11:53:20 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on brewing/brewpiless/mode: b'o'
2022-01-16 11:53:20 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on brewing/brewpiless/pressure: b'0.0'
**2022-01-16 11:53:20 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on brewing/brewpiless/fridgeHumdity: b'='**
2022-01-16 11:53:35 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on brewing/brewpiless/state: b'1'
2022-01-16 11:53:35 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on brewing/brewpiless/beerTemp: b'22.9'
2022-01-16 11:53:35 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on brewing/brewpiless/beerSet: b'21.0'
2022-01-16 11:53:35 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on brewing/brewpiless/fridgeTemp: b'23.4'
2022-01-16 11:53:35 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on brewing/brewpiless/fridgeSet: b'23.0'
2022-01-16 11:53:35 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on brewing/brewpiless/roomTemp: b'24.7'
2022-01-16 11:53:35 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on brewing/brewpiless/mode: b'o'
2022-01-16 11:53:35 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on brewing/brewpiless/pressure: b'0.0'
**2022-01-16 11:53:35 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on brewing/brewpiless/fridgeHumdity: b'>'**