vitotai/BrewPiLess #353: Recalibrating Gravity due to the effects of co2 and pressure

shingle01, 2021-12-09:
I wish I had a bigger brain or a degree in fluid mechanics.  

We can clearly see that pressure is increasing the specific gravity/buoyancy of the iSpindel.

**Nice to have:**

- iSpindel Beer Temperature correction
- iSpindel vessel pressure correction.

[image]

vitotai, 2021-12-10:
Temperature correction is already implemented.

I don't have any idea about pressure correction and how to handle it.

shingle01, 2021-12-10:
Thanks I recall seeing the temperature correction, sorry for missing.

I think there are two additional factors.  The density increase due to co2 absorption and the effect of pressure on the ispindel.  I observed the gravity continue to 

Warning - armchair physics.

My ipsindel saw a SG increase of 0.005.  This is presumably while fermentation is occurring and SG is actually decreasing.

Using the following formula we can estimate the mass of co2 being absorbed by the beer as pressure increases:
http://braukaiser.com/wiki/index.php?title=Carbonation_Tables

Cbeer = (Phead+1.013)*(2.71828182845904^(-10.73797+(2617.25/(Tbeer+273.15))))*10

- Cbeer - carbonation of the beer in g/l
- Phead - head pressure in bar
- Tbeer - temperature of the beer in C

Lets assume I was pretty close to Phead of 1bar (14.5 psi).  Tbeer at 18C

Cbeer 	= (Phead+1.013)*(2.71828182845904^(-10.73797+(2617.25/(Tbeer+273.15))))*10
	= (1+1.013)*(2.71828182845904^(-10.73797+(2617.25/(18+273.15))))*10
	= 1.013*(2.71828182845904^(-10.73797+(2617.25/291.15)))*10
        = 1.013*(2.71828182845904^-1.748617)*10
	= 1.762766 g/l
        = 1762.766 m/l

if Specific Gravity = Density of sample / Density of pure water

and If the following additional weight affect Specific Gravity
10,000 mg/l 1.01
20,000 mg/l 1.02
50,000 mg/l 1.05
100,000 mg/l 1.10
150,000 mg/l 1.15

Therefore a sample of pure water under 1bar of pressure at 18C would have a specific gravity of 1.0018 once an equilibrium has been reached.

It would appear co2 absorption doesn't account for the full increase but must be contributing.

shingle01, 2021-12-10:
As i'm publishing the data to mqtt and picking up in home assistant, I have created a calibrated gravity attribute with the formala above.  For pressure I am using a statistics sensor with a 6hr sliding window, this should roughly allow for equalibrium.  

For pressure, I'm going to do some testing.

Measure the tilt with instantaneous co2 pressures in water and high gravity liquid.  This will rule out co2 absorption as a factor and create a second offset.  Unfortunately, these results will be purely anecdotal but I suspect I will observe a linear result that may be easy to estimate.

vitotai, 2021-12-11:
I've done a few pressure fermentation, or Spunding. I haven't seen such gravity difference.

BTW, someone has mentioned temperature correction of the acceleration sensor is also needed. I choose to ignore it for now. There are so many causes of errors that I don't think we should pursue this kind of precision.

shingle01, 2021-12-11:
The accelerometer MPU 6050 has onboard temperature correction.  What additional temperature compensation is required?

No problem, I'll continue to augment in Home Assistant.

closed 2021-12-11 by shingle01