vitotai/BrewPiLess #354: Translated DisplayLcd.cpp and invalid state on index.html page

petobena, 2021-12-18:
Hi @vitotai ,

I have translated strings in DisplayLcd.cpp but now I see only "Invalid state" on index.html page
The strings on Oled LCD are working
Where can be problem?
Thank you

[image]
[DisplayLcd.txt](https://github.com/vitotai/BrewPiLess/files/7740037/DisplayLcd.txt)

vitotai, 2021-12-21:
The web page parses the displayed string on LCD to get state and time, if any.
The "state", like "waiting to cool", is in fact available for now, but there is no convenient way to get time string.

petobena, 2021-12-21:
so now is there no way to translate the "state" ?
is there any chance to use strings from locale file in future? make sense to rewrite the code?

vitotai, 2021-12-21:
Well, it is now done in latest 4.2pre branch. (no pre-built image for now.)
It's what I should've done when I changed classic HTML display.

petobena, 2021-12-22:
its now perfect, thank you! :)

closed 2021-12-22 by petobena