vitotai/BrewPiLess #356: 4.2 Humidity Sensor

shingle01, 2022-01-09:
Hello.  Testing 4.2pre.  Mostly all good.  Two issues with humidity sensors:

Defining a DHT22 chamber humidity sensor doesn't commit.  I click apply, refresh and the definition is lost.  I've tried re-init all devices and starting from scratch with no success.

When defining a room humidity sensor the new web GUI does not give an option to specify device type e.g. DHT11/22 etc

vitotai, 2022-01-09:
I mess up options of Humidity control, BME280, and DHT support. They were all default disabled.
Try building the latest push.

shingle01, 2022-01-10:
My DHT22 chamber sensor is now working.  Thanks.

Though for completeness, adding a Room Humidity sensor doesn't enable the 'sensor type' dropdown.

[image]