vitotai/BrewPiLess #358: Stability Problems

kmycek, 2022-02-02:
Hi, Vito. I'm really struggling to find an existing build, or a build a version myself that is stable on a Wemos D1 Pro. I've tried a two different D1 Pros, and I've tried builds from 3.6 all the way to 4.2. The later versions seem to have more stuff working - like backups and restores - but have issues with local logging (space?). With any version past 3.6, I cannot modify anything on the config screen (title, username and password, etc). But the most troubling is the stability. After a few hours of working properly, the config will randomly disappear (reverting back to the initial config - no installed devices, or WiFi creds, etc). Do you think it is a D1 problem? Would I be better off with an ESP32?I mean I know that those D1 minis are cheap and probably 40% of them are DOA - but I'm not sure that would explain why the config is evaporating after working properly for a few hours. I'm going to flash ESPHome on them and see if I can just operate my hardware with Home Assistant - to prove that there's nothing wrong with the hardware. 

The new stuff is great - pressure transducers and humidity, but not if the cost is at the expense of local logging and stability. At the end of the day, I need a bit more confidence that this software is stable and won't crash catastrophically during a brew. I could deal with loosing WiFi occasionally - but I would be heartbroken if I came back after a day or so and my beer was free rising because the controller reverted back to a new install. 

Feel free to email me directly at kmycek at yahoo dot com if you have any suggestions or need anymore data. I'm a C# programmer by trade and I could probably dive in with a little guidance.

quimnut, 2022-02-02:
You can format the spiffs to see if that fixes stability issues.
http://brewpiless/format-spiffs

vitotai, 2022-02-03:
try @quimnut 's suggestion.

Check flash ID and real size on system page.
I bought 4 ESP32. One of them has unstable network issue. One other worked fine until it didn't. No clue at all.
I am running latest under-verifying-but-available-on-github 4.2 on my newly built ESP32. It works as expected.
Before this new ESP32 build, my ESP8266 at 4.1 worked since it was available.

kmycek, 2022-02-06:
Not sure how to decipher the values on the the system page (and since I reverted back to v3.6, I don't even see a system page - it's still a "config" page - but I do remember that from when I had a later version. 

v3.6 seems to be pretty solid for a few days now, and I have local logging capabilities. The fridge I have it deployed in is all buttoned up, and I don't want to take it all apart, again. I'll try the format-spiffs on another test version I have in my office. 

But long story short, the format-spiffs seemed to fix it. This issue can be closed. Strange that it needs to be done on a fresh flash. Maybe put that in the WIKI?

closed 2022-02-06 by kmycek