vitotai/BrewPiLess #361: D1 R2 not supported anymore?

b-brewing, 2022-03-15:
I've had brepiless for a while now on 2 D1 R2's. If i'm not mistaken they are v3.6. Lately i've been having some issues and wanted to upgrade to latest but can't seem to get it to work on any later versions. Should I pay attention to something specific or is it simply not supported?

kmycek, 2022-03-15:
If by D1 R2, you are talking about the Wemos D1 Mini (Pro, with the external antenna) - then I too have not been able to run anything past v3.6 reliably, and with all the core functionality, like logging. If I need to go up to an ESP32, so be it - but I will have to redesign my PCB.

b-brewing, 2022-03-15:
D1 R2 is a regular sized arduino board. I just tried a few builds and the only one i get to work is the "classic" on v3.6. 
One of the issues i was having in this one is that eventually it will not be able to log anymore. It won't show any past logs and won't let me start a new one.

vitotai, 2022-04-02:
The memory is so tiny that local logging is impossible. I couldn't fit the new images, neither.