vitotai/BrewPiLess #362: Rapt Pill

Liero91, 2022-04-01:
Hello,
is there a plan to implement the new Kegland Rapt Pill Hydrometer?

Thank you!

Lukas

vitotai, 2022-04-02:
I don't know much much of it. It works only with RAPT server, doesn't it?

Scales82, 2022-04-02:
Yeah, as far as I know, its closed-source for now. I was at the Kegland facility last weekend and discussed with the boss about it very briefly. I wouldn't count on it being let-loose anytime soon. They are concentrating hard on their own line of products.

Scales82, 2022-04-19:
https://www.youtube.com/watch?v=Rg-0JppKOEA

This video just popped up. Could this webhook technique on the RAPT portal be possibly tweaked to send to a BPL instead of Brewfather in this instance?

Scales82, 2022-04-21:
Ok guys, I've had success linking the RAPT pill to BPL.

Using the video mentioned above, I used this payload below and got BPL to register inputs as if it were from an iSpindel.
Must keep calc using BPL off for it to work.

```
{
  "name": "iSpindel000",
  "ID": "1234567",
  "angle": "10",
  "temperature": @temperature,
  "temp_units": "C",
  "battery": @battery,
  "gravity": @gravity,
  "interval":900,
  "RSSI": @rssi
}
```

The only tricky thing (for me anyway) is getting the webhook from the RAPT portal to be delivered to my BPL on my LAN securely. As I'm only amateur at those things I'm not sure the most secure way to do it. I tested it with a basic port-forward on my router and it worked fine. I know there are webhook relays etc but they seem to be paid services generally for a lot of forwards.

Any ideas to get this across the line?

shingle01, 2022-04-22:
Nicely done sir.  

If you have a Home Assistant box perhaps you could create a webhook trigger automation that forwards the json data on to brewpiless?

e.g. set rapt.io webhook url to https://your-home-assistant:8123/api/webhook/rapt-pill-webhook-relay-id123

Create Home Assistant automations:

- id: '1650592899681'
  alias: Rapt Pill Webhook Relay
  description: ''
  trigger:
  - platform: webhook
    webhook_id: rapt-pill-webhook-relay-id123
  condition: []
  action:
  - data_template:
      data: '{{ trigger.json }}'
      url: 192.168.1.4
    service: shell_command.forwardjson
  mode: single

Scales82, 2022-04-23:
Would that be insecure having my HASS exposed to the greater internet? if not, could I just not expose my BPL via a port forward and enable password security on it?

shingle01, 2022-04-23:
HASS more secure in my opinion.

HASS will support ssl and webhook ids.  HASS GUI security is more robust.

Scales82, 2022-04-23:
fair enough. I just dont have HASS set up yet. was hoping for a simpler solution than going down that rabbit-hole.
preferably something that can run on a docker stack quickly and simply

lbussy, 2022-04-23:
This is the reason I didn't consider this for BPR.  Opening your home firewall to outside connections is dangerous. If you think simply using a single internal target is safe you are mistaken. 

Anytime I say this people argue:

- They know what they are doing.  They do not.  If they did they would not suggest this.
- They have nothing that would interest a hacker.  If you ever use or connect to that network with any systems with which you do any online transactions, you do.  The transactions need not happen on that network, they only need to connect to it once after with a system that does. 
- They only open one port and you can't do anything with that.  That's wishful thinking and I see professionals leverage any connection.  
- No hacker would be interested in hacking their network.  It doesn't matter, they hack anything they can.  Bots identify weaknesses of anything connected to the Internet. That weakness gets logged and categorized with thousands of others and then more specific bots compromise thousands at a time.

I'm just some dude on the Internet. What do I know?  I manage teams that support sensitive systems and 70% of our efforts go to Cyber.  I see what our side does to test our systems with relative ease.  I can see what would happen if those attempts were malicious. I can also see how much work
It takes to secure a system and I know no one person can do that for their home environment.

So, do what you will, but in the world climate today you should only do it if you also think having unprotected sex with a Haitian hemophiliac IV drug using prostitute is fine.

Scales82, 2022-04-25:
Was there a suggestion there? no offence but that was all very good information but does not help out the goal here right?

lbussy, 2022-04-25:
Yes, I thought it was clear.  Until they allow pulling the data, just do without.  When I spoke to them a few weeks ago they were not fully against the idea, they just were concerned about the additional server load.

If you want an affirmative answer: Create a web service on an Internet-facing server, and then use SOAP or other pull methods to retrieve the data on demand.

While OS developers are not responsible for the choices their users make, I strongly feel OS should not ignore common sense and just leave it all to the consumer who does not fully understand the implications.

shingle01, 2022-04-26:
HASS is designed to be Internet-facing.  Ibussy needs to chill.  With that said I thought I'd implement my solution:

The following worked for me.  I had to break out the json payload into individual topics as HASS was removing the quotation marks messing with the syntax Brewpiless expected.  

**configuration.yaml**

shell_command:
  forwardjson: >-
    curl -X POST -H 'Content-Type: application/json' -d '{"name": "iSpindel000", "ID": "1234567", "temperature": {{temperature}}, "temp_units": "C", "battery": {{battery}}, "gravity": {{gravity}}, "RSSI": {{RSSI}}}' {{ url }}

**automation.yaml**

- id: '1234567890'
  alias: Rapt Pill Webhook Relay
  description: ''
  trigger:
  - platform: webhook
    webhook_id: your-webhook-id
  condition: []
  action:
  - data_template:
      name: '{{ trigger.json.name }}'
      ID: '{{ trigger.json.ID }}'
      temperature: '{{ trigger.json.temperature }}'
      temp_units: '{{ trigger.json.temp_units }}'
      battery: '{{ trigger.json.battery }}'
      gravity: '{{ trigger.json.gravity }}'
      RSSI: '{{ trigger.json.RSSI }}'
      url: http://192.168.4.1/gravity
    service: shell_command.forwardjson
  mode: single

**Rapt webhook payload**

{
  "name": "iSpindel000",
  "ID": "1234567",
  "temperature": @temperature,
  "temp_units": "C",
  "battery": @battery,
  "gravity": @gravity,
  "RSSI": @rssi
}

**Rapt webhook url** 

https://your-home-assistant:8123/api/webhook/your-webhook-id

lbussy, 2022-04-26:
Sure, it will be fine. I'm sure [nobody has ever hacked it](https://www.home-assistant.io/blog/2021/01/22/security-disclosure/).

tappyson, 2022-09-18:
Hi All 

I have Home Assistant absorbing data from the RAPT cloud via the API. I want to somehow send this data locally to BrewPiLess. Haven't managed to figure anything out from the above. Clearly too dumb.

andycwk, 2022-11-17:
I'm sure I read in the RAPT stuff at some point that the pill (as well as the temp prob they are now producing) will connect to the RAPT fridge and ferm controller via BT...

.. This leads me to thinking that the Pill is probably advertising it's info via a BT beacon like the Tilt does.

As a new owner of a Pill, I'm going to be looking into this a bit deeper - if it does offer a beacon then it should be relatively easy to extend BPL to consume the Pill like it now does for the Tilt!

vitotai, 2024-01-16:
I just make it work to support Pill via Bluetooth.