vitotai/BrewPiLess #363: Beer Profiles are broken in v4.2p3

alexlark, 2022-04-07:
Issue: Saved beer profiles are truncated and hyperlinks broken in v4.2p3

To replicate issue: Create a beer profile and save it, navigate away from the beer profiles, navigate back to beer profiles and load a beer profile. The saved profiles will be truncated and hyperlinks broken.

Issue does not exist in v4.1r4

vitotai, 2022-04-07:
I just push a new version, modified layout. Try if it works for you.

alexlark, 2022-04-07:
I can see the new DEL buttons, and they work, but the profile names are still truncated and do not work :-/

[image]

vitotai, 2022-04-08:
what browers are you using?

[image]
[image]

alexlark, 2022-04-08:
I'm using Chrome and I tried in Edge as well.

The profiles save and I can click them but once I navigate away from the page, and back, they are then truncated. Also, my profiles do not have a a forward slash prefix and when I navigate to http://brewpiless.local:8008/filemanager I am unable to see them saved.

The DEL button removes them but when navigating away, and back, they re-appear.

Version v4.1r4 is OK

[image]

vitotai, 2022-04-09:
I didn't change anything about control panel or profiles after 4.1. In fact, I totally forget profile loading/saving. There was a delete buttons in classic UI, but it didn't show up in new UI. It was until you mentioned that I just knew that. 

What on my Mac with Chrome:
[image]

vitotai, 2022-04-09:
It's difficult for me to solve issues if I couldn't reproduce by myself. However, if you can provide more information, I might be able to give it a try.
On Chrome, select "View"->"Development Option"->"Javascript Console".
Click the icon marked as "1" in the following picture. Click on around "DEL" button. then Click "Computed" marked as "2". Capture the screen and show it to me.
[image]

alexlark, 2022-04-09:
This has got to be linked to the way the list of profiles are fetched. The list of profiles are being cut by 2 characters so all further GET /POST requests are using the wrong URL....

[image]

But the profiles are actually getting saved in the correct location....

[image]

vitotai, 2022-04-10:
Yeah. You are right. 
I'm converting to LittleFS and didn't notice the different behaviors between LittleFS and SPIFFS.
Just push a new commit. 
Sorry about that.

alexlark, 2022-04-10:
All working now, thanks. Loading and deleting work as expected.

Just a minor observation, when viewing the filemanager the profiles folder does not expand whereas the logs folder does. Just pointing this out if this was unintended or helps with the LittleFS conversion...

[image]

closed 2022-04-22 by alexlark