vitotai/BrewPiLess #365: Translation!

Neca-Grof, 2022-04-16:
Hello Everyone! 
I would like to translate the complete program into Serbian, all commands and the complete web interface, as well as messages on the LCD screen. 
I use a windows computer. can someone explain to me how to translate ??
Thanks!

dumpamob, 2022-04-22:
https://github.com/vitotai/BrewPiLess/wiki/Localization-of-web-pages

Neca-Grof, 2022-04-24:
I use windows os, so I don't have the ability to change the language, footer and header ... I use ms visual studio code, when I click build, upload, there are no changes, I'm still in English. any solution?

dumpamob, 2022-04-24:
_...just translate the json file and send it to me (Vito Tai). (Or, submit an issue with attached translated file, I guess.)._

Translate the file and attach it here (dragging & dropping). Check @vitotai that soon he will make an update generating the translation, it's not just creating the file, you have to generate the language files.

vitotai, 2022-04-29:
Surely it works for me, on Mac.

Neca-Grof, 2022-04-30:
oh, I don't have mac, I use windows computer :(

Neca-Grof, 2022-04-30:
> _...just translate the json file and send it to me (Vito Tai). (Or, submit an issue with attached translated file, I guess.)._
>
> Translate the file and attach it here (dragging & dropping). Check [...]

I would like to learn the procedure of generating the language files myself! :)

vitotai, 2022-04-30:
Follow the directions in https://github.com/vitotai/BrewPiLess/wiki/Localization-of-web-pages

If you successfully generate the html files in dist directory, you can upload it to verify.
The last step is just to generate header files so that the htmls are embedded.

Neca-Grof, 2022-05-01:
I worked exactly according to that instruction! the wakanda folder has appeared and contains html files. when I click on the bug icon in vs code, then on esp8266 - bulid -upload, there is no change in the language. I'm wrong somewhere.

vitotai, 2022-05-01:
It's uploading by file manager.
Try browse to "http://[your BPL IP]:8008/filemanager
and upload the gzipped html file.

Neca-Grof, 2022-05-03:
Thanks! I do that, but it don't work... still on english...