vitotai/BrewPiLess #368: iSpindel gravity readings are not plotted in 4.2

tomicheck, 2022-05-06:
Hi,

I've just upgraded to 4.2 (BrewPiLess.v4.2.rc2.oled.esp8266.4m2m.bin) and I can see iSpindel gravity readings on OLED:
 
[image]

and on testcmd.htm:

[image]
 
but they are missing in the graph:

[image]

vitotai, 2022-05-07:
The value of plato was considered as invalid value, which is wrong and corrected.

IDMaynard, 2024-01-12:
I am unable to get the iSpindel data to show up on the graph, in version 4.2.  It appears on the LCD and on the side of the graph but is not getting plotted.  Any suggestions?
!
[image]