vitotai/BrewPiLess #370: Logged data is lost on power cycle

Pete-in-Perth, 2022-08-04:
Hi. I've just upgraded to BPL 4.2 (was something like 3.2 previously). I often use the data logging function. In the new version, it appears data is lost after a power cycle.
I'm using a Wemos D1 Mini board with Thorax configuration.
Have attached a screenshot.
Any suggestions?

[image]
.

vitotai, 2022-08-07:
There is a new option, "Write on Buffer", which should be NO/FALSE/DISABLED to encourage writing whenever there is new data.

[image]

Pete-in-Perth, 2022-08-16:
Thanks for your reply. I've played around with the "Write on Buffer Full" setting. 

With it turned on, the log is lost completely on a power cycle, and a new log needs to be started.

With it turned off, the log appears to be there but there is no data showing from before the power cycle (as per the screenshot on my first post).

Thanks!

vitotai, 2022-08-26:
I tried it on my testing platform, an ESP32, but I couldn't reproduce it.
However, something strange happened on my working setup in my last brew. The log stop like someone stop it while I didn't remember doing so. I am still hurting this issue.

Adilson6060, 2022-09-06:
It is the same with me . It was right , but I don’t known why brewpiless font show the fermentation begin #
[image]

dumpamob, 2022-09-26:
It happened here too, if you turn off the power, the graph disappears in the interface and also in the log when viewed by the Log Viewer.

bandaxadibar, 2023-02-01:
I confirm the feedbacks informed by the other people. Yesterday, all the data disapear.

gumazza, 2023-02-18:
I'm using the latest version (4.2rc4) and the problem still persists, here after 5 days the LOG disappears.
[image]

ksga, 2023-05-10:
Running on Wemos D1 Mini's, and had the same issue with logs being lost on power cycles.
Grabbed the source this morning, made the same changes as @havardAasen committed, and can confirm that it appears to work:
[image]

danestead, 2024-01-09:
I'm having this same issue, though I'm on version 4.3.

I've looked at havardAasen's edit to brewlogger.cpp but the source looks the same, so I haven't made any changes.

Anyone else having this problem and able to provide a solution?

danestead, 2024-01-10:
> Thanks for your reply. I've played around with the "Write on Buffer Full" setting.
>
> With it turned on, the log is lost completely on a power cycle, and a new log needs to be started.
>
> With it turned [...]

Hi Pete, I'm also in Perth. Did you manage to fix this?

danestead, 2024-01-10:
> Running on Wemos D1 Mini's, and had the same issue with logs being lost on power cycles. Grabbed the source this morning, made the same changes as @havardAasen committed, and can confirm that it [...]

Is this still working for you?

Regarding the change to brewlogger.cpp, did you manually type in the changes or download the whole file and replace?

dumpamob, 2024-01-11:
here with 4.3.1, D1 Mini, and making the changes suggested by the @havardAasen. solved. 
I deleted the lines that the @havardAasen marked, saved, build and upload.

danestead, 2024-01-11:
> here with 4.3.1, D1 Mini, and making the changes suggested by the @havardAasen. solved. I deleted the lines that the @havardAasen marked, saved, build and upload.

Ah, so delete the lines shaded pink/red?

dumpamob, 2024-01-11:
> > here with 4.3.1, D1 Mini, and making the changes suggested by the @havardAasen. solved. I deleted the lines that the @havardAasen marked, saved, build and upload.
> 
> Ah, so delete the lines shaded pink/red?

Yes that's right.

PR #393 (2024-01-16): Flush data to flash on ESP8266 and ESP32
When we write logs, we need to flush the data to flash, so we don't lose it. This was not possible on ESP8266 because of definitions to only allow ESP32 access to these functions. This restriction is now removed.

This change should fix #370
closed 2024-01-16 by vitotai