vitotai/BrewPiLess #372: Relay module continues to work, does not trip.

Leidge, 2022-09-02:
Something very strange is happening here.
I used the 2 channel relay module.
BrewPiLess, starts to work perfectly.
Sometimes it works perfect for days.
But sometimes it turns off and the Relay continues to work, it doesn't disarm. So the temperature continues to drop.
Has anyone ever experienced this.
BrewPiLess reports that the fridge is inactive. But the relay still works.

[image]

ithinkido, 2022-09-02:
You could try 2 x 10k pull down resistors
[image]

Leidge, 2022-09-02:
It would be 2 resistors of 10K.
One in GND and in In 1, and other in GND and In 2?

lbussy, 2022-09-02:
My money is on a bad relay.  These being "rated" at 10A is a generous number.

Remove the relay completely, test the continuity.  If the relay shows closed, it's a bad relay.

Leidge, 2022-09-02:
Then. I've tested it with 3 relays. All new.
The same happens.
When I install new, it stays about 4 days without any problem.
But then it has the same problem.

lbussy, 2022-09-02:
Then you are overloading the relays.  Look for "Locked Rotor Amps" on the compressor and see what that says.  Add 50% again and that's the relay rating you need.  You might consider SSRs.

Hugo-Pereira, 2022-09-21:
I also have a similar issue. Sometimes the fridge/heater won't turn on. Shaking the box I have the components on sometimes makes it work...until it stops working again.
I have almost zero knownledge on electronics.
I checked my connections and all seems fine. Re-did all the wiring, problem came back.
Could it be because we're powering the relay with the 5V from the arduino board? My arduino board itself is powered through USB.
I've ordered a new relay, will replace and see if that fixes the issue.

vitotai, 2022-09-21:
I have some relay modules that work well while some don't.

Please remember the GPIO of ESP8266 is rated at 3.3v. While 3.3v is considered as HIGH in 5V logic, some "bad" relay modules just don't work that way. 
Please also make sure to provide enough current, at least 2A.