vitotai/BrewPiLess #383: Please could you add a remotely activated  ESP.restart() function

Hwerow, 2023-01-09:
Hi Vito,
I am using the MQTT function to display information and I find sometimes that I need to restart the ESP32 to get the MQTT to work.
Rather than having to go to the brewery, to restart the BPL, I would like to be able to remotely restart the ESP32 - ESP.restart()
Perhaps using Other URLs a suitable Restart command could be sent?
Thanks

vitotai, 2023-01-16:
The real problem is unreliable MQTT connection. I checked the source of AsyncMqttClient library and found that relying on connection status is not enough. Checking publishing result is necessary to for confirming the connection is still available. I've made the change. Please try if it works better.
As to the restart function, the function is already there and just adding a small piece of code, it's done. A "/restart" path is also added.
You will have to build the source by yourself.

Hwerow, 2023-01-16:
Thanks very much - I will check it out
Cheers
Peter

closed 2023-01-16 by Hwerow