vitotai/BrewPiLess #385: Unable to export to CSV using BPL LogViewer v3.0

pedro-jorge-pereira, 2023-01-16:
I can open the attached log file but cannot export it to CVS, using the BPL LogViewer v3.0, available at [https://github.com/vitotai/BrewPiLess/tree/master/htmljs/dist/english](https://github.com/vitotai/BrewPiLess/tree/master/htmljs/dist/english)
[APA0622.log](https://github.com/vitotai/BrewPiLess/files/10426037/APA0622.log)

vitotai, 2023-01-17:
Found and solved. Try the latest viewer.

pedro-jorge-pereira, 2023-01-17:
Thank you, Vitotai, for the quick fix.

Sadly, I can't still export my log file. Do you know if your fix is compatible with old log files?

I saw that your commit does not modify the BPLLogViewer.htm.

Could you open and try to export the log file? I am doing the following steps without luck:

1) download from GitHub the master branch.
2) unzip the master branch.
3) navigate to ..\BrewPiLess-master\htmljs\dist\english  folder.
4) open the BPLLogViewer.htm file on chrome.
5) click to open file dialogue, and select my log file.
6) click export.

vitotai, 2023-01-17:
sorry. I used a wrong command. Please try again. Or here is the exported file:
[APA0622.log.csv](https://github.com/vitotai/BrewPiLess/files/10434001/APA0622.log.csv)

pedro-jorge-pereira, 2023-01-17:
@vitotai I do confirm that now it is working.
Thank you.

closed 2023-01-17 by pedro-jorge-pereira