vitotai/BrewPiLess #391: Glitchy temperature readings

glenosAU, 2023-02-10:
I find I get inconsistent temperature readings from my BrewPiLess. It is like it is dropping data packets, I notice it with the room temp probe (grey line). This then seen to affect the reading for the fridge temp.

It was fine until I rebuilt my set up into a bigger box using three pin connectors for the temp probes. Clearly it seems related to that change. The only difference is the connectors for temp probes and the power supply, instead of a wall plug transformer I have a PSU in the case.

Anyone got a tip on where to look? Or should I rebuild it?

[image]
[image]

alexlark, 2023-02-13:
My money would be on the PSU. I had an unreliable PSU like that style in the past. Now using a RPi PSU.

glenosAU, 2023-02-15:
Thanks for the suggestion, I'll give it a try when I next brew. Do you
think it would be worthwhile to increase the voltage output to 5.2V or so?

alexlark, 2023-02-15:
I analysed the output on the PSU with a multimeter and it appeared erratic, +/- 0.5v as such I just didn't trust it, especially adjusting the pot to increase voltage. I assumed that type if PSU may be more suited to LED lights etc. I figured as the RPi PSU was designed for the Pi it would be more stable, the output on the multimeter proved that and it's been rock solid for over 3 years. After using the RPi PSU for a few weeks I stripped it down (the wall wart) so it would fit in my enclosure.

quimnut, 2023-02-15:
You could try put a 470uF smoothing capacitor across the 5v input.

glenosAU, 2023-02-22:
Thanks for the suggestions. I was ordering some parts for my coffee machine from mouser and need to spend a bit more for free delivery. So I got one of these.
https://au.mouser.com/ProductDetail/CUI-Inc/PSK-15D-5-T?qs=DRkmTr78QAQphYGnb4sRWQ%3D%3D

Testing so far, 1 hour, shows a lovely stable signal.

closed 2023-02-22 by glenosAU