vitotai/BrewPiLess #400: Both sensors reporting beer temp

Hugo-Pereira, 2023-05-25:
On my latest brew, I'm having the weirdest bug. After some time, brewpi starts registering the fridge temperature being the same as beer temperature. This is causing the heater to run way longer than it should, having my fermentation chamber reached more than 60ºC!
My wiring seems correct, and the brewpi bahaves correctly for hours, until randomly it decides to stop reading the fridge temperature.

See image attached. I'm on version 4.2. What could be causing the issue?
[image]

Thanks in advance

Daniel-Magg, 2023-06-06:
I have seen the same issue. Power cycling seems to fix the problem.
Would the system stopped heating/cooling if the chamber sensor is either stuck or cannot be read.

Hugo-Pereira, 2023-06-25:
I changed the resistence on the temperature probes to 2k2 and the issue hasn't happened again on my tests. I'll report again how it goes on my next brew

Daniel-Magg, 2023-06-29:
Ok, thanks, I might try that.

It happened to me yesterday again and I have observed that in the device list that value for the chamber temperature sensor was blank (after refreshing the device list). So it should be possible to detect this condition in SW and disable the control system.
I'm using the Thorrak PCBs (surface mount main PCB and RJ-45 breakout board).
My configuration is:
Device 0, device slot 0, HW type sensor, function Beer Temp
Device 1, device slot 1, HW type sensor, function Chamber Temp
Device 2, device slot 2, HW type sensor, function Room Temp
Device 3, device slot 3, HW type pin (inverted), function Chamber Cooler
Device 4, device slot 4, HW type pin (inverted), function Chamber Heater

Hugo-Pereira, 2023-06-29:
Yes I've also seen that on the device list.
Personally I'd love an option to reboot automatically. 
E.g. if we're logging, and the device has been on for more than 1 hour, and we lost one of the probes, reboot.
Rebooting always fixed the issue, but only temporarily.

Lets hope the 2k2 fixes it though. I also added a capacitor on the probes for good measure.

Hugo-Pereira, 2023-07-04:
It's been 5 days and I've got no issues to report. Seems that the 2k2 resistor solved it! (Or the capacitor, or both)