vitotai/BrewPiLess #404: Device Page / Refresh Device List Not Working

JonathanThorpe, 2023-08-03:
Hi All,

I have a D1 Mini Pro running BPL 4.3 which seems to have problems with the "Refresh Device List".

I have tried:
1. Formatting the filesystem.
2. "Init / Erase EEPROM" doesn't help, but with the Serial Console open, I see lots of:
`ERROR: Too many messages queued`

Which I understand comes from the ESP Async WebServer.

When I look at the browser console, I see lots of references to:
`Uncaught SyntaxError: missing ] after element list`

Trying to trace this, it appears to be related to the way that the embedded JS in setup.html is evaling the output from the Websocket:
`for (var key in eval("m={" + msg + "}"),`

Looking at testcmd, I can see a lot of these messages and I suspect it's because some messages start with a "," - e.g:

```
20:58:20 ↓ A:{"rssi":-55,"st":0,"md":"o","bt":-20000,"bs":-20000,"ft":-20000,"fs":-20000,"rt":-20000,"sl":"Idling for 22m37","tu":"C","ptc":"o","pt":0,"ptctp":0,"ptclo":0,"ptcup":400,"pm":0,"psi":0,"G":{"u":0,"t":-25000,"r":-999,"g":-1000}}
20:58:26 ↓ A:{"rssi":-55,"st":0,"md":"o","bt":-20000,"bs":-20000,"ft":-20000,"fs":-20000,"rt":-20000,"sl":"Idling for 22m43","tu":"C","ptc":"o","pt":0,"ptctp":0,"ptclo":0,"ptcup":400,"pm":0,"psi":0,"G":{"u":0,"t":-25000,"r":-999,"g":-1000}}
20:58:32 ↓ A:{"rssi":-55,"st":0,"md":"o","bt":-20000,"bs":-20000,"ft":-20000,"fs":-20000,"rt":-20000,"sl":"Idling for 22m49","tu":"C","ptc":"o","pt":0,"ptctp":0,"ptclo":0,"ptcup":400,"pm":0,"psi":0,"G":{"u":0,"t":-25000,"r":-999,"g":-1000}}
20:58:33 ↓ N:{"v":"0.2.4","n":0,"c":"00000000","s":0,"y":0,"b":"e","l":"1"}
20:58:35 ↓ d:[]
20:58:35 ↓ h:[D:{"logType":"I","logID":10,"V":[-1]}
20:58:35 ↓ {"i":-1,"t":0,"c":1,"b":0,"f":0,"h":1,"d":0,"p":13,"x":1}D:{"logType":"I","logID":10,"V":[-1]}
20:58:35 ↓ ,{"i":-1,"t":0,"c":1,"b":0,"f":0,"h":1,"d":0,"p":14,"x":1}D:{"logType":"I","logID":10,"V":[-1]}
20:58:35 ↓ ,{"i":-1,"t":0,"c":1,"b":0,"f":0,"h":1,"d":0,"p":15,"x":1}D:{"logType":"I","logID":10,"V":[-1]}
20:58:35 ↓ ,{"i":-1,"t":0,"c":1,"b":0,"f":0,"h":5,"d":0,"p":-1,"j": 0.000}]
20:58:38 ↓ A:{"rssi":-55,"st":0,"md":"o","bt":-20000,"bs":-20000,"ft":-20000,"fs":-20000,"rt":-20000,"sl":"Idling for 22m55","tu":"C","ptc":"o","pt":0,"ptctp":0,"ptclo":0,"ptcup":400,"pm":0,"psi":0,"G":{"u":0,"t":-25000,"r":-999,"g":-1000}}
20:58:44 ↓ A:{"rssi":-55,"st":0,"md":"o","bt":-20000,"bs":-20000,"ft":-20000,"fs":-20000,"rt":-20000,"sl":"Idling for 23m01","tu":"C","ptc":"o","pt":0,"ptctp":0,"ptclo":0,"ptcup":400,"pm":0,"psi":0,"G":{"u":0,"t":-25000,"r":-999,"g":-1000}}
20:58:50 
```

Is there something I've missed?

Thanks,
Jonathan

allthepies, 2023-08-03:
When I look at my websocket messages I get the "h" command all on one line, whereas your commands output shows the h command split over multiple lines, that's presumably why the comma error is occurring.  Also, my h command doesn't have the D:{"logType".... entry.  Here's mine:

JonathanThorpe, 2023-08-10:
I think we can possibly put this down to my environment and the dependencies it was pulling down for building the code. I ended up trying BrewFlasher to load the 4.3 release and it works fine. It seems faster and has no errors at all.

gumazza, 2024-06-08:
It also reboots when I try to access the devices page.
But this only happens when the circuit is powered by an external source, because when the power comes from the USB connected to the PC, everything works normally.