vitotai/BrewPiLess #408: Heating/cooling control only with beer sensor

jacyker, 2023-09-26:
Hi 

Beer on kveiks. In this case, it is not about cooling, but about heating. I use a heating belt on the fermenter. The problem is that it doesn't work when there is only a beer sensor (DS18B20 or iSpindel). BPL also requires a "fridge sensor", but with kveiks it is often a shame to block the refrigerator. You can add a second DS18B20 "in the room" and set it as a "fridge", but it would probably be better if in such a situation the BPL simply followed the beer sensor only, with appropriate "caution"?

vitotai, 2023-11-20:
There is an option, Glycol support. You can use that specific binary.

vitotai, 2023-12-15:
I just update a new version for making glycol an option, so that you can install all available versions and set the "glycol chilling" configuration in system page.

jacyker, 2023-12-16:
OK, so only the beer sensor and the "Glycol Chilling" option. Thanks for the info and update :-)

jacyker, 2023-12-16:
:-)

closed 2023-12-16 by jacyker