vitotai/BrewPiLess #410: ESP32 will not detect DS18B20

BartPichola, 2024-01-04:
I have an issue with 4.3.1 not detecting DS18B20 on ESP32 board. It will not pull up the sensor on any pin. 
- I have tried multiple known good probes. 
- Sensor works fine on ESP8266 board with v4.3.1
- Sensor works fine on the same ESP32 board with a simple OneWire Arduino sketch
- DS18B20 sensor is wired in normal mode (not parasite)
- Voltage checks good (3.3v) 
- 4.7K resistor checks good
Any ideas?

vitotai, 2024-01-04:
My ESP32 working setup works fine.
Which PIN do you use?

BartPichola, 2024-01-04:
I've tried all available pins. Most testing was done on pin 27.
I've ordered new probes and a couple of new ESP32 boards. I'll test when they arrive and I'll report back.

vitotai, 2024-01-05:
The default OneWirePin is 23. 
I bet you know what you are doing, so good luck.
BTW, make sure your sensors are not parasite mode. BrewPi, and hence BrewPiLess, doesn't support parasite mode sensors. Google for more information.

BartPichola, 2024-01-06:
ok, I got a set of new probes and they work. BrewPi detected right away. I'm assuming that all 3 of the probes I had are either defective or not built to spec. Turns out to be a hardware issue. Thanks for your help.

closed 2024-01-06 by BartPichola