vitotai/BrewPiLess #415: BEER PROFILE displays Centigrade values/charts when set to Fahrenheit mode.

jsemroc, 2024-02-08:
I'm experiencing an issue with BrewPiLess and Beer Profiles when running in Fahrenheit mode (v4.3 & v4.3.1 running on ESP32-WROOM-32D).

After setting BrewPiLess to Fahrenheit mode (via SYSTEM --> ADVANCED), I create a Beer Profile on the CONTROL page and set a start date of "Now", SAVE and then APPLY.
[image]
If I then navigate to any of the other pages (GRAPH, LOGS, DEVICE, etc) and then return to the CONTROL page, the Profile column heading indicates degrees "F" yet the temperature values as well as the Y-axis are reported in degrees "C".
[image]
The only way to get the Profile to reflect Fahrenheit values again has been dozens of page-refreshes or randomly browsing the other tabs (GRAPH, LOGS, DEVICE, SYSTEM, GRAVITY, PRESSURE).

vitotai, 2024-02-11:
Yeah, there are issues about using Fahrenheit, but it's not like yours.

Anyway, I've modified the code that might have something to do with you issue.

Remember, I am still testing/verifying this version 4.3

jsemroc, 2024-02-12:
I compiled the 4.4 code today and installed onto my ESP32. Unfortunately, now the CONTROL page never reflects the switch to Fahrenheit visually or on the profile's graph. In case it was really operating in Fahrenheit mode, I tried creating a profile with 66F values but when I visit the GRAPH it show the profile set point is 150.8F so it appears that CONTROL is not switching once the ESP32 has been configured for Fahrenheit mode.

vitotai, 2024-02-13:
Your issue might be caused by weak/incomplete communication between your computer/phone/pad and the controller.

It might because of
1. configuration issue.
2. Hardware issue
3. software issue.

All I can do is about software issue, and I've done my best. I would suggest 
1. try reset/clear the whole flash of your ESP32.
2. try another ESP32