vitotai/BrewPiLess #417: 2 Button with ESP32

glenosAU, 2024-02-19:
Hi,
Can anyone advise the code changes need to enable 2 button use on the ESP32 board? The instructions in the Docs is for the D1 Pro board and the pin assignments are different.
https://github.com/vitotai/BrewPiLess/blob/master/docs/10.HardwareGeneral.md#input-options

I tried fiddling in Config.h and different pin combinations to no result.

Thanks
Glen

vitotai, 2024-02-20:
The only pins available of ESP8266 are D0 and D2 that should be left open when boot. That's why I implemented two buttons. The operation is awkward in my experience.  ESP32 has more than enough pins for rotary encoder, so rotary encoder is the only supported input device.

glenosAU, 2024-04-17:
OK.
I asked as I was retrofitting an ESP32 to a previous brewpiless build which used an ESP8266.

closed 2024-04-17 by glenosAU