vitotai/BrewPiLess #418: Beer profile date changes after Apply (in v4.4, esp32 with glycol and pill)

drbubbles, 2024-02-23:
After Applying a Beer Profile, the dates in the profile settings change to ~1970 (but not quite the unix epoch, a bit later).  The profile appears to start correctly, so it is working. Here are before and after screenshots for a simple case.  This is for version 4.4, on an esp32 with glycol enabled, and sensing SG from a pill. 

[image]

[image]

vitotai, 2024-02-24:
Fixed.
I forgot to convert EPOC time for Javascript Date constructor when loading profile, which is times 1000.
You can upload "control.htm.gz" instead of flashing new image if the fermentation is on going.

drbubbles, 2024-02-24:
I tried uploading "control.htm.gz" using "restore" in the "System" tab in BrewPiLess, but that must be the wrong way. How do I upload it? (apologies for my ignorance!)

vitotai, 2024-02-27:
you will need to go to http://[your controller IP]:8008/filemanager

to upload the file.

closed 2024-05-13 by drbubbles