vitotai/BrewPiLess #422: Device setup page calibration only allows integer value, but JsonCommand allows float

denisfrench, 2024-03-10:
Hi; I've noticed on the device setup page (http://brewpiless.local/setup.htm) that when float calibration values are entered, they are converted to an integer. I've only tested with DS18B20s.

However, I am able to enter float a float calibration value using a JSON command (https://github.com/vitotai/BrewPiLess/blob/691a92b89435d3a1124f79776f9185f83488bcc3/doc/JsonCommand.md#sensor-calibration). This does work as a float, and is even then displayed as a float on the device setup page.

Perhaps there is something up with the Javascript that only allows a float value entry? I'm compiling version 4.4 ce1428821f12e2aa687425625735fb9e7f9b4627 using the ```esp32-oled``` default build environment and config.

Thanks for your work on this!

vitotai, 2024-03-17:
found and fixed. Thank you.

denisfrench, 2024-03-21:
Confirmed fixed at my end; thanks! 🙏

closed 2024-03-21 by denisfrench