vitotai/BrewPiLess #423: Cannot setup DHT22 Room Humidity Sensor, but setup as Chamber Humidity Sensor works

denisfrench, 2024-03-10:
Hi; I've noticed on the device setup page (http://brewpiless.local/setup.htm) that I cannot setup a DHT22 as a Room Humidity Sensor in the first instance. The device stays when I hit "apply", but the sensor value shows 255 continiously, and the temperature sensor does not show up when devices are refreshed.

I am able to work around this by setting the DHT22 as a Chamber Humidity Sensor first, and then change it to a Room Humidity Sensor. This then works okay, but the sensor continues to be reported in MQTT as ```fridgeHumidity``` not ```roomHumidity```.

I'm compiling version 4.4 https://github.com/vitotai/BrewPiLess/commit/ce1428821f12e2aa687425625735fb9e7f9b4627 using the ```esp32-oled``` default build environment and config.

Thanks for your work on this!

vitotai, 2024-03-17:
I missed a few lines of code for Room humidity sensor which I never tested. Sorry for that.

denisfrench, 2024-03-21:
Confirmed fixed at my end; thanks! 🙏

denisfrench, 2024-03-21:
I was too quick!

Adding DHT22 as room sensor now works, except it is still being reported within MQTT as ```fridgeHumidity```.

Before this fix using the workaround it was reporting a value, even through the topic was incorrect. Now it is reporting random single keyboard symbols (eg ,*-.).

Nearly there!

vitotai, 2024-03-29:
Yeah, the tag was wrong. However, I don't get it how the value was wrong.

denisfrench, 2024-03-29:
Hi; is the latest commit (b05b25a4809ac255990bb0366f5019efd97b34ab) ready for testing?

I'm getting an error when compiling ```esp32-oled``` as follows:
```
src/MqttRemoteControl.cpp: In member function 'void MqttRemoteControl::_reportData()':
src/MqttRemoteControl.cpp:165:59: error: expected primary-expression before ',' token
             if(rh <= 100) lastID=_publish(KeyRoomHumidity,,rh);
                                                           ^
*** [.pio\build\esp32-oled\src\MqttRemoteControl.cpp.o] Error 1
================================================================================================== [FAILED] Took 20.17 seconds ==================================================================================================
Environment    Status    Duration
-------------  --------  ------------
esp32-oled     FAILED    00:00:20.166
```

koekie, 2024-04-08:
I fixed it in the following way:
`if(rh <= 100) lastID=_publish(KeyRoomHumidity, rh, 0);`
(When you leave out the `0` the `_publish` method assumes the value is a `char`, which it is isn't.

I have fixed some of the issues already in my forked version of this repo (for which I might prepare a PR for in the future if @vitotai is up for that, but I will reach out when I have the time to do so), check the following:
https://github.com/koekie/BrewPiLess/blob/master/src/MqttRemoteControl.cpp#L162-L171

denisfrench, 2024-04-21:
@vitotai release version 4.4 (https://github.com/vitotai/BrewPiLess/commit/a37150fe4a5f7133d463e3fc8708bed54a1140c1) fixes the roomHumidity topic, but the MQTT data now randomly sends either ```"``` or ```!``` for some reason.

@koekie your latest commit (198ab1339e3f14395a2c712c4382bfa95d7126a7) fixes this issue in toto, thanks!