vitotai/BrewPiLess #424: "Humidity" and "Room RH" toggles work, but don't change appearance like the other toggles

denisfrench, 2024-03-10:
Hi; I've noticed on the main graph page (http://brewpiless.local/index.htm) that the "Humidity" and "Room RH" toggle switches work to disable or enable the graph display of these values, but don't change appearance to show on/off like the other toggles do.

It's only a cosmetic issue, but I just thought I would let you know.
 I'm compiling version 4.4 https://github.com/vitotai/BrewPiLess/commit/ce1428821f12e2aa687425625735fb9e7f9b4627 using the ```esp32-oled``` default build environment and config.

Thanks for your work on this!

vitotai, 2024-03-17:
This issue is caused by using the same ID for two elements. Fixed.

denisfrench, 2024-03-21:
Confirmed fixed at my end; thanks! 🙏

closed 2024-03-21 by denisfrench