vitotai/BrewPiLess #425: Temperature last v 4.4 version

WojtekGD, 2024-05-05:
Hi
In the last version it displays the angle nicely but shows the Spindel temperature - 0 degrees.
 A similar problem is when entering Original Gravity and Specific Gravity - converted to 0 degrees.
 The temperature measurement from the Spindel was always displayed in the devices tab and draw in yellow on the graph as Aux Temp.
[image]
[image]

vitotai, 2024-05-24:
Mine seems to work fine.
Would you please keep the page opened, open Developer/Javascript Console, and wait for iSpindel to report. The controller will relay the messages from iSpindel to the browser and log to Javascript console. Copy the message for me, so that I can figure out what goes wrong.

WojtekGD, 2024-05-28:
Hi.
I'm sorry my mistake. BPL is working properly, my spindel has failed. Exactly Dallas is damaged. It quickly drained the battery and the spindel stopped working altogether. Yesterday I finished fermentation and removed the spindel, which allowed me to diagnose the problem.