vitotai/BrewPiLess #429: Free rise option in beer profile

eyebinker, 2024-06-02:
Would it be possible to add a free rise option in the beer profile tab.  For instance on a Belgian beer have the 1st step in its profile ignore the beer temp unit it raises to a specific temp, then activate temperature control.

Hwerow, 2024-06-11:
That would be very useful,  as would inhibiting heating when cold crashing.

quimnut, 2024-10-10:
I would use this, I disable the heating pin currently.

vitotai, 2024-10-21:
You could just chill the wort to, say, 18C, and set the temperature to the target of your free rising, like 24C. That's what I've been doing.

I don't believe in "free rise" thingy because I live in a place where the temperature is higher than 30C half of the year and over 25C most of the time. If I let my fermentation rise freely, it would rise to 25C in a few hours, usually before lag time elapses.

Hwerow, 2024-10-21:
I understand your situation. However, it would be a nice option to have if it wouldn't be too much trouble?