vitotai/BrewPiLess #432: MQTT configuration.

raul811, 2024-09-10:
I use this application for my fridge and I want to send MQ.T message to my home assistance.
I tried everything to set up the MQT server. But I don't have any topics received in my home assistant..

It is possible to verify the logs for the MQ. TT in the bruises, or how can I do?

raul811, 2024-10-10:
In the configuration of bru piles, I need to use individual topics or Jason.

I see in the configuration from Homo asdistant U use 304 time and the topic is brupaillas.

Thank you very much for your kind and sorry.If I disturb the the discussion in rather topic I delete it all the message.Thank you again

denisfrench, 2024-10-10:
[image]

raul811, 2024-10-10:
[image]

raul811, 2024-10-10:
This is my configuration from blue pilots

denisfrench, 2024-10-10:
You need to toggle the switch next to "Reporting" on.

raul811, 2024-10-10:
The problem was.I forgot to check the report

Now it's okay and I thank you very much for your help.

raul811, 2024-10-10:
[image]
[image]

closed 2024-10-17 by raul811