vitotai/BrewPiLess #435: [FR] Update MQTT reporting on change of state, mode, beerSet or fridgeSet

denisfrench, 2024-10-10:
I have successfully setup capture of BrewPiLess data within Home Assistant via MQTT.

Control via MQTT is also working, but I have found the lack of immediate feedback makes programmatic MQTT control problematic. 

Please consider updating MQTT reporting on change of ```mode```, ```beerSet``` or ```fridgeSet```, to complement the MQTT remote control functionality. Update of ```state``` upon change would also be useful for monitoring purposes, as it often changes within the ```Publish Period``` timeframe.

Likely this is relevant to MQTT control through other approaches, such as NodeRed, Python etc.

Thanks for your work on this 🙂

raul811, 2024-10-10:
Hello, guys.
I try.
 To connect my blue pilots via MQT to home assistant by but I don't know how can I do it?

What is the simple method to connect to empty teeth?Broke out like mosquito in homo assistant

What is the simple method to connect to empty teeth? Broke out like mosquito in Homo assistant.

Sorry for my bad english.
 And thank you very much.

denisfrench, 2024-10-10:
@raul811; please don't hijack this thread with off topic discussion.

There is no simple method. It requires manual configuration. Welcome to Home Assistant!

All the information you need is here https://www.home-assistant.io/integrations/sensor.mqtt/ , and here
https://www.vito.tw/BrewPiLess/I0.MQTT.html .

Use https://mqtt-explorer.com/ to learn, experiment, test, and troubleshoot.

If you have a BrewPiLess specific issue that looks like a bug or documentation problem, please open your own issue.

raul811, 2024-10-10:
Sorry, but not this was my question.

I try to configure my brupilenyes but I don't know what are the best or true configuration in log tag

I try to configure my brupilenyes, but I don't know what are the best or true configuration in log tag?

I know in home assistant.Is another topic for anotget hoget up?But the homes incident is working.Okay.Because I have another m.Q.T clients and it's all right
I know in home assistant. Is another topic for anotget hoget up? But the homes incident is working. Okay. Because I have another m. Q. T clients, and it's all right.
Thank you. You're sorry again for the question.

raul811, 2024-10-10:
Sorry again for my bad english. And how can I write this? Because I'm a blind guy and I'm trying to do something here and I need some help.

denisfrench, 2024-10-10:
@raul811; please let's continue this conversation on your issue #432 .

https://docs.github.com/en/site-policy/acceptable-use-policies/github-disrupting-the-experience-of-other-users

glenosAU, 2024-10-19:
I have used your example from #432 and got brewpiless via MQTT into Home Assistant. The only control I can make work is to set the brew mode but I cannot set the temperature for fridge or beer constant.

Got any tips?

denisfrench, 2024-10-19:
Are you certain it is not working? The key issue is that feedback via MQTT is delayed, therefore you will not see it update in HA until the next ```Publish Period```. Check the BrewPiLess webif directly for real-time updates.

For this reason I have given up on control from HA, hence raising this FR.

The YAML from https://github.com/vitotai/BrewPiLess/issues/432#issuecomment-2404552570 was otherwise working fine for me. Probalby also check you have the full topic (ie it does not automatically include ```Base Path```) in your "MQTT Remote Control" settings (https://github.com/vitotai/BrewPiLess/issues/432#issuecomment-2404603519), which had me stumped for a while.

Concievably we could reduce ```Publish Period``` to 5 or 15 seconds and it would probalby work, but I would rather not spam my MQTT server, and wonder about the load on my BrewPiLess which already spontaniously resets occaisionally.

glenosAU, 2024-10-19:
I have made some progress. Setting the `Publish Period` to 15s helps when trying things out.
I have set up configuration.yaml using your code from #432 and the MQTT example from https://www.home-assistant.io/integrations/input_number/#automation-examples

configuration.yaml: (I commented out the Pill bits as I don't use one.
```
# brewpiless MQTT
mqtt:
  sensor:
  # BrewPiLess
    - name: "Brew State"
      unique_id: "brew.state"
      state_topic: "brewpiless/state"
      value_template: >-
        {% set mapper =  {
          '0' : 'Idle',
          '1' : 'Off',
          '2' : 'Door Open',
          '3' : 'Heating',
          '4' : 'Cooling',
          '5' : 'Waiting to Cool',
          '6' : 'Waiting to Heat',
          '7' : 'Waiting for Peak',
          '8' : 'Cooling Min Time',
          '9' : 'Heating Min Time' } %}
        {% set state = value %}
        {{ mapper[state] if state in mapper else 'Unknown' }}
      icon: mdi:state-machine
      expire_after: 301 # BrewPiless MQTT Publish Period set to 300s
    - name: "Brew Fridge Temp"
      unique_id: "brew.fridge.temp"
      state_topic: "brewpiless/fridgeTemp"
      unit_of_measurement: "°C"
      device_class: temperature
      expire_after: 301 # BrewPiless MQTT Publish Period set to 300s
    - name: "Brew Room Temp"
      unique_id: "brew.room.temp"
      state_topic: "brewpiless/roomTemp"
      unit_of_measurement: "°C"
      device_class: temperature
      expire_after: 301 # BrewPiless MQTT Publish Period set to 300s
    - name: "Brew Temp"
      unique_id: "brew.beer.temp"
      state_topic: "brewpiless/beerTemp"
      unit_of_measurement: "°C"
      device_class: temperature
      expire_after: 301 # BrewPiless MQTT Publish Period set to 300s
#    - name: "Brew Pill Temp"
#      state_topic: "brewpiless/auxTemp"
#      unit_of_measurement: "°C"
#      device_class: temperature
#      expire_after: 301 # BrewPiless MQTT Publish Period set to 300s
#    - name: "Brew Pill Gravity"
#      state_topic: "brewpiless/gravity"
#      icon: mdi:beer
#      state_class: measurement
      # device_class: specific_gravity # Not yet a thing!
#      suggested_display_precision: 3
#      expire_after: 301 # BrewPiless MQTT Publish Period set to 300s
#    - name: "Brew Pill Battery"
#      state_topic: "brewpiless/voltage"
#      unit_of_measurement: "%"
#      device_class: battery
#      suggested_display_precision: 0
#      expire_after: 301 # BrewPiless MQTT Publish Period set to 300s
  number:
  # BrewPiLess
    - name: "Brew Fridge Temp Set"
      unique_id: "brew.fridge.set"
      state_topic: "brewpiless/fridgeSet"
      command_topic: "brewpiless/fridgeControl"
[... 123 lines cut ...]
        state: Fridge Constant
    card:
      type: tile
      entity: input_number.fridgeset
      icon: mdi:fridge
      features:
        - type: numeric-input
          style: buttons
  - type: conditional
    conditions:
      - condition: state
        entity: select.brew_mode
        state: Beer Constant
    card:
      type: tile
      entity: input_number.beerset
      icon: mdi:glass-mug-variant
      features:
        - type: numeric-input
          style: buttons
```
In non-edit mode the Fridge Temperature Setting tile is not visible in Beer Constant mode.
[image]

Are you brewing anything interesting? I pitched K-97 yeast onto an Altbier today.

glenosAU, 2024-10-30:
I've found a related problem. If I set the beer constant temperature through HA and check that it has been applied in Brewpiless it works as expected. Even with a 300s MQTT update time.

The strange behaviour is at some time later it will change to 18°C. I thought it was happening due to poor wifi signal which I fixed a few days ago.

For example, yesterday I set the beer constant through HA to 4°C at 4pm. At 1:55am today the set temp changed to 18°C. The previous se point was 8°C so it isn't reverting to a previous set point.

When I set the set point through Brewpiless it stays as set.

Attached picture is HA history for yesterday and today. The gap yesterday was me adjusting the WiFi extender.
[image]